dresden-elektronik/deconz-rest-plugin #2456: Support request for BTicino DIN power consumption module

meco489, 2020-02-16:
Hi, I have installed a RaspBee for my home automation.
I can connect some BTicino products, using Legrand integration .... under the list of functioning products.
I can't connect a DIN module for power measurement.
I attach below the screenshots.
Thanks in advance and congratulations for the work you are doing.
Domenico

**Working Products:**
 - [BTicino K4003C](https://catalogo.bticino.it/BTI-K4003C-IT) equal to Legrand 067773
 - [BTicino  3584C](https://catalogo.bticino.it/BTI-3584C-IT) equal to Legrand 064888
 - [BTicino  K4531C](https://catalogo.bticino.it/BTI-K4531C-IT) equal to Legrand 067775

**Non-functioning products:**
 - [BTicino F20T60A](https://catalogo.bticino.it/BTI-F20T60A-IT) 

[image]

[image]

[image]

SwoopX, 2020-02-16:
Could you be a bit more precise on what non-functional means? I'd assume you don't see it in Phoscon (which is kind of expected) and you don't have any power sensor and readings?

Happy to fiddle in the few lines to add it as supported device somewhere the next days.

@Smanar Legrande is you domain as I read ;)

meco489, 2020-02-16:
Right .... I can't associate it in Phoscon ...
Sorry: I took it for granted even if it wasn't!
Thanks

Smanar, 2020-02-17:
@SwoopX , lol, yep, I can make this integration, all the code is already in place, will be fast.

@meco489 
Can you take the same screenshoot but with consumption ? I ask that because for exemple on plug legrand use Voltage + Current + Power, but in reality only Power is working.
The device is out of the box, or you already have updated it with bticino gateway ? (legrand electrical mesurement don't work without update)
And if I make code modification, do you know how to compile it to test ?

meco489, 2020-02-17:
Hi, when I made the screeshot that I sent you the BTicino module was connected .... I also saw that the active power was 0 .. but surely the reading was about 800W ??

P.S.
On the BTicino APP it only displays the active power.

The device is updated with the latest firmware available.
Is the compilation the same as described here? https://github.com/Smanar/deconz-rest-plugin/blob/master/README.md

Smanar, 2020-02-17:
>But surely the reading was about 800W ??

So you can confirm me, at a moment you have a value at 0x050B attribute ?

And, yep, it's less or more this link, you have the official one at deconz github (it 's same than on my fork) https://github.com/dresden-elektronik/deconz-rest-plugin and you just need to change an url.

Give me an hour, I will give you a new link. I need to delete the previous one and remake one.

Smanar, 2020-02-17:
Ok done, so you can follow the procedure "I**nstall deCONZ development package (optional, Linux only)**"
But with using this command at step 1:
`git clone https://github.com/Smanar/deconz-rest-plugin.git`

I haven't compiled it yet, my production machine is busy, but the code is so easy, I don't think I have made error.

I haven't found your device id (0x000d) So for the moment I imagined a new one "DIN power measurement" (will see later)

meco489, 2020-02-17:
Super fast !!!

After following your instructions here is what I managed to do:

- I managed to associate the instrument with Phoscon APP and on deCONZ-GUI I read the active power.

[image]

- On the integration of Home Assistant I see the tool and I can have the instantaneous active power

[image]

- Strangely, on Phoscon App I don't see the tool anywhere ...

I have achieved the goal of integration with Home Assistant .... I am at your disposal to carry out further tests to improve the Plug-in.

Let me know what I can do for you!

Thanks so much!

espen4001, 2020-02-18:
Hi! I have a similiar case. I have a smart relay, which has alot of data shown in deconz, but only on/off switch in Phoscon. Can you help me get all features available to home assistant? I created another issue, [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2464](url)
Looks like you know what to do, maybe you can help me? @Smanar

Smanar, 2020-02-18:
@espen4001 Don't worry @SwoopX did more integration than me ^^.

But your issue need same screenshoot than this one (at minimum basic cluster, to be sure we are on good device + cluster list + electrical maesurement cluster for create a device for it)

@meco489 Yep, legrand is easy, all the code is already on place in deconz, so it's fast.
And yes not all device can be show in Phoscon, but this app is closed source, so we can't do nothing for that.
I will try to find what is the device type 0x000d, but If this device don't have more feature, I will PR the code as it.
Thx too for adding a new device in legrand branch ^^, we have same device too here.

SwoopX, 2020-02-18:
Don't worry, I got the Develco stuff covered, ven without screenshots. The technical documentation is really good ;)

SwoopX, 2020-02-19:
@meco489 Which deconz version are you currently running? If it's ok for you, could you update to .73? I'm curious to see if that 0x000d device ID thingy has been resolved. Made some additions in that regard for this particular version.

meco489, 2020-02-20:
@SwoopX   I'm stuck ....

> 
> -------------------------------------------------------------------
> 
>              marthoc/deconz Firmware Flashing Script
> 
>                        Version: 0.5
> 
> -------------------------------------------------------------------
> 
> 
> Listing attached devices...
> 
> /usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)
> 
> Enter the full device path, or press Enter now to exit.
> 
> Device Path : /dev/ttyAMA0
> 
> -------------------------------------------------------------------
> 
> Firmware available for flashing:
> 
> deCONZ_ConBeeII_0x26490700.bin.GCF
> deCONZ_ConBeeII_0x264a0700.bin.GCF
> deCONZ_Rpi_0x26330500.bin.GCF
> 
> Enter the firmware file name from above, including extension,
> or press Enter now to exit.
> 
> File Name : /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
> 
> -------------------------------------------------------------------
> 
> Device: /dev/ttyAMA0
> 
> Firmware File: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
> 
> Are the above device and firmware values correct?
> Enter Y to proceed, any other entry to exit: y
> 
> Flashing...
> 
> /usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)
> 
> Flashing Error! Please re-run this script...

... some idea?

SwoopX, 2020-02-20:
Hm, I got no clue about the docker stuff... However, from what I can see from the screenshots, you're running a deconz version which is at least half a year old. Which version is it?

I've heard marthoc/deconz releases follow pretty quick after a new deconz release. Maybe on on .73 is already available? Hope there's also some guidance on migration.

meco489, 2020-02-21:
I was a little behind ..... solved!

Device ID OK!

[image]

If it's okay for you at this point I would report it as closed ....

SwoopX, 2020-02-21:
Sweet. I'd prefer to leave it open as I'd expect that the electrical measurement data is not available yet unless you haven't compiled my repo.

meco489, 2020-02-21:
The only electrical measurement that displays the device, even with the original app, is the instantaneous active power .... I read it even earlier in Home Assistant.
Have you changed anything?

SwoopX, 2020-02-21:
> The only electrical measurement that displays the device, even with the original app, is the instantaneous active power .... I read it even earlier in Home Assistant.
> Have you changed anything?

I basically copied Smanar's changes. So from one of your earlier posts, I'm not sure if you compiled and tried his version. Anyway, if you also get reasonable readings in HA, then perfect. Fine to close the request then.

meco489, 2020-02-23:
Sorry maybe I was not exhaustive .... I compiled and tried the version of Smanar: the reading of the active power in HA was precise.

Thanks again for your work.

closed 2020-02-23 by meco489

erbagiovanni, 2020-03-05:
Hi all!
@meco489 how di you manage to see Power readings? In my HA the device is seen as light and obviously has no attributes

SwoopX, 2020-03-05:
You may need to delete the device and re-pair it.

SwoopX, 2020-03-06:
Maybe this?

> @SwoopX , lol, yep, I can make this integration, all the code is already in place, will be fast.
> 
> @meco489
> Can you take the same screenshoot but with consumption effective ? I ask that because for exemple on plug legrand use Voltage + Current + Power, but in reality only Power is working (other value stay at 0)
> The device is out of the box, or you already have updated it with bticino gateway ? (legrand electrical mesurement don't work without update)
> And if I make code modification, do you know how to compile it to test ?

meco489, 2020-03-06:
**@erbagiovanni** 
Hi, in phoscon I have never seen it .... it was associated with phoscon but only displayed on deconz-gui.
In HA it was automatically recognized as a current sensor.
Good luck

Smanar, 2020-03-06:
And @SwoopX is right, on my side I have never see a Legrand device works for measurement without update.
IDK for bticino, but are you using an updated device or one "out of the box" ?

And yes sensor is invisible on Phoscon.

erbagiovanni, 2020-03-06:
Maybe this Is what I am missing. What you mean with "updated"? I see firmware version and It says 35.

Smanar, 2020-03-06:
So you have the same than @meco489 , no reason for yours don't work.
This device can create 2 devices, 1 light because it's a wired device (but useless) and a sensor device.

Are you sure you have nothing news in HASS  ?

Dou you know ho to acess to sensors list using JSON and webbrowser ?
With an url like this one > http://deconz_ip:deconz_port/api/api_key/sensors

SwoopX, 2020-03-06:
A deconz restart (and HASS as well) also sometimes helps.

seymourddr, 2020-03-06:
I have the same problem. DIN module with firmware 35 is integrated in deconz but is not seen in phoscon or in hassio.

SwoopX, 2020-03-07:
@erbagiovanni @seymourddr just to close the obvious out. You are both running on .74, right?

erbagiovanni, 2020-03-07:
Yes, .74. Tomorrow I'll try to get sensor list via JSON and put it here.
I'll also try ti see the component in deconz GUI

Smanar, 2020-03-07:
Yep, thx.
And if you can compare that you see in the GUI with the screenshoot on previous post, to see is something is different,pls

erbagiovanni, 2020-03-07:
Done, here what i see, via JSON I see it under lights e not under sensors

[image]

[image]

Smanar, 2020-03-07:
Ok, so it's because the last version.
The device called "Misuratore potenza assorbita" is the conbee.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2525
Long story, but during your inclusion pahase, was not the device that was detected but the conbee.

The biticione device is the one that start with 000474

And ATMit's unreachable on your screen, the name is grayed.

and the pairing process have failed because you have as name the nwk 0x482f

erbagiovanni, 2020-03-07:
Ok, if I understand well from other threads there we'll be a fix in next days. Many thanks for your support!

Smanar, 2020-03-07:
The fix will be for the name problem, not for the device integration.
For me the pairing have missed, if it work you will see the node with yellow bar connected to the node with blue bar (the conbee).

Better with ungrayed name
More better with correct name "Consumption awarness device"

I havent this device to test, but it haven"t led to check process ?

seymourddr, 2020-03-07:
> @erbagiovanni @seymourddr solo per chiudere l'ovvio. Entrambi sono in esecuzione su .74, giusto?

yes
[image]
[image]

erbagiovanni, 2020-03-07:
Question: the device is already associated with the legrand gateway. Could this be the problem?

Smanar, 2020-03-07:
Nope, because you need to reset it, to pair it with the conbee.

erbagiovanni, 2020-03-07:
ok, so you can confirm it's forbidden to have une device paired to more than one network (in my case one with legrand GW, the other with Conbee)? Sorry for all this questions but I'm new to this world. 
If so, is there any chance to solve my problem (avoiding to buy a second device)?

SwoopX, 2020-03-07:
@erbagiovanni It's not forbidden, it's simply not possible afaik. It is pretty much like with wifis. The wifi ID is the PAN ID in zigbee and the network key which as a similar function is zigbee. The exception is that a device can only remeber the gateway it is currently associated with. Otherwise, it must be reset and paired again.

That's probably also the reason why you haven't any line drawn in your screenshot.

erbagiovanni, 2020-03-08:
Now it's clear to me. All my BTicino devices are integrated in HA via "integrations" functionality and legrand gateway, but for some reason this module is not seen there. So, thanks again all for tour support, I'll find another way!

Smanar, 2020-03-08:
But you are using the official Legrand gateway or the conbee ?
And you wan to use wich one ?

erbagiovanni, 2020-03-09:
At the moment I have both. I would like to keep official Legrand to let my family use their own app (easier...) and conbee to control with HA everything, legrand devices and other sensors. This mixed scenario is ok for all devices except this power module, as even if connected to legrand gateway it's not shown in HA (while all lights and power outlets are correctly mapped)

rotragit, 2020-03-23:
HI all, OK, I'm trying to pair the bTicino F20T60A module. The module is joined to the network but cannot understand how to read the values in Home Assistant. In HA integrations tab, scrolling the list of the devices paired with the ConBeeII I cannot find anything.

I'm running deconz in docker, version .75. I'm sure I'm missing something very stupid, I don't find it.
Can you help me?

[image]

Smanar, 2020-03-23:
Hello, I think you device is bad paired.
The name is grayed and (but I can't be sure yet) if all is good you will have a "name" instead of 0x4e17.

If all is good you will see a new sensor (not light) in HA.

rotragit, 2020-03-23:
Thank you. I think there is something wrong with the database, because I've
reset the F20T60A module (red light is always on) and I deleted the node in
the deConz GUI. But if I restart deConz the node is still there.

Can I check the database in any way?
Thank you.
romano

rotragit, 2020-03-23:
Ok, found on the db:

sqlite> select * from devices;
...
17|00:04:74:00:00:a1:32:f0|1584857930|19991

but it is not in nodes nor in sensors. Thereis a "lights" table? I don't
see it with .tables......

thank you
r.

ebaauw, 2020-03-23:
`nodes` is the lights table, i.e. the table containing the `/lights` resources.  `devices` contains the nodes as shown in the GUI; `device_gui` contains the screen coordinates, and `device_descriptors` contains the simple descriptor per endpoint (as a BLOB).  These are linked by to `devices` by `id`; `nodes` and `sensors` are linked by `mac`, which is the prefix for `uniqueid` (which is called `mac` in `nodes`).

Smanar, 2020-03-23:
@rotragit don't loose your time trying to delete it
Just re-include it, will be faster, and I think you will need somes other tries ^^.

The node is stll here but I think the NKW (the 0xXXXX name) have changed ?

rotragit, 2020-03-24:
@ebaauw @Smanar  thank you. I tried several times but it's only paired as a light (and router). I have cleaned the db now, it's a 3 minutes work or less. 
Just not to pair in the wrong mode: I have to pair in deConz or in Phoscon? In the latter, as a sensor, switch or light?

SwoopX, 2020-03-24:
It must be Phoscon. Should not really matter, what you use for the search.

rotragit, 2020-03-24:
OK, this is the situation at the moment. In the db it's listed in devices but not in sensors nor in nodes.
[image]

If I delete the node, reset the bTicino module and pair again, the result is the same.
Should I expect to find a line in sensors?

Smanar, 2020-03-24:
Now it's fine, don't touch your network ^^.

And yes, even it don't work, you need to have a new sensor.

Can you show the JSON of your device, the light one, and if you find it the sensor one.

There is many way to have itn but the easier way is using a web browser

http://IP:PORT/api/API_KEY/lights/
http://IP:PORT/api/API_KEY/sensors/

You have few device, so it will be easy to see them (or it if the sensor not working).

If you select the second circle in the node, you have "Consumption awareness device" ?

rotragit, 2020-03-24:
Yes, if I select the second circle in the node, I have "Consumption awareness device".
[image]

Smanar, 2020-03-24:
Have you used phoscon for inclusion ?

You can force the API device creation with a tips
- set phoscon in permit join
-  in the Node Info panel, read the Basic cluster attributes

And ofc without reseting/deleting the device.

rotragit, 2020-03-24:
Bingo!

Yes, I used phoscon for inclusion. With your tips now I see the sensor also in HA.

(Thank you)E1000 :-)

Smanar, 2020-03-24:
NP, IDK what happened, but it s like the device was included but directly with deconz.
Perhaps you have a too big permit join duration on it ?

SwoopX, 2020-03-24:
@Smanar Just out of curiosity, is there a space before the manufacturer and modelID? I had a quite similar behavior just a couple of days back while trying to have the sensors for a Huawei device created. Reading some basic attributes didn't help, so I had to implement a rather exotic change...

@rotragit Glad it finally worked out.

Smanar, 2020-03-24:
@SwoopX heyyyy, You are right (good eyes again ^^), I m looking in wireshark there is 0x20 before the string for both and I think all legrand device

But It work when I do 
>if (sensor->modelId() == QLatin1String("Dimmer switch w/o neutral") ) then ...

Perhaps some functions trim the string (but not all)  ?

SwoopX, 2020-03-24:
Hm, good question. It's an assumption.

This is the only place I could spot any impact on the defined modelID at the beginning of de_web_plugin.cpp (first glimpse). And on the bindings...

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/ca72c46fd147dcf5c0b04304a96316d2c97ff477/de_web_plugin.cpp#L7476-L7498

However, it might be that you're right and the strings get trimmed.

rotragit, 2020-03-25:
@Smanar , the first ten attempts was done from Phoscon for sure, because only after about ten failures I have investigated on how connect to deconz inside the docker container (I had done some experiments the month before trying to export DISPLAY and running the program from outside but without success, then I understand how to connect using VNC, etc etc).
For sure one or two attempts was done trying to join into deconz directly. But I'm sure the last attempt (and a few before it) was done again in Phoscon because I removed from the db the entry, and I asked what is the correct way to pair the module and you told me: Phoscon.
In the dconz the max time to permit join is 254 second. I never put it "always open".

Anyway, I have other two modules like that (I've got three for 25 euro each :-) ). So today I will try with one just "out of the box" and I let you know if the problem is the same...

[239 more replies]