dresden-elektronik/deconz-rest-plugin #2458: Cant get my deConz online.

najklez, 2020-02-16:
Hi
Am a total novice in this subject but have done some awsom work with Hassio and conbee2.
After I moved hassio from Rpi to Oracle VM on PC windows7 I got problems. Tried to connect to Hue Essential app on android. No success. Tried to re install deConz. That's where all problems started. DeConz cant connect. 
[image]
[image]
  #####[image]

najklez, 2020-02-17:
Now im sure im using the rigt string, /dev/ttyACM0. it will not start if i use something else, 

[image]

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix