dresden-elektronik/deconz-rest-plugin #2460: add support for iris motion sensor v3

redwoodimage, 2020-02-16:
Manufacturer: iMagic by GreatStar
Product Name: Device
Model Number: 1117-S
deviceTypeId: 15
manufacturer:iMagic by GreatStar
address64bit:D0CF5EFFFEC8D68B

Here's a links with more info on the sensor [https://community.hubitat.com/t/new-iris-motion-sensor-version-generation-3/5190/2](url)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

rmotapar, 2020-02-21:
Just wanted to say that I am seeing the same issue as well when adding iris motion sensor v3. It shows up in the mesh network but does not show up in the webapp. I do see an open pull request  for this so that's good news!

[image]

SwoopX, 2020-02-21:
The device will not show up in Phoscon, as the software is closed source. DE must handle integration there.

rmotapar, 2020-02-21:
@SwoopX  Thank you! I assumed it would show up in Phoscon app. Are there any other ways to get this device integrated with home assistant when using Conbee ii?

SwoopX, 2020-02-21:
When the new deconz version gets release, the missing sensors should be created automatically and therefore show up in HA. If not, just try to repair.

Meanwhile, you could add the missing bindings manually for power configuration, temperature and humidity cluster. Regarding motion, I'm not sure if the IAS Zone cluster is in enrolled status. If not, you can also enroll it manually by using the MAC address of your Conbee. Search the issues for binding guidance (its really not that difficult) and for manual enrollment, that has been explained in one of the heiman CO2 sensor issues.

rmotapar, 2020-03-02:
@SwoopX  Thank you for the response! I was able to add the iris motion sensor v3. Temperature and humidity values started showing up in homeassistant after pairing but as you suspected there was nothing related to motion. Thanks for the pointers on how to enroll them manually, I was able to enroll the IAS Zone cluster and motion events are now showing up. Thanks again! 

I am wondering what it would take to do the same for Iris motion sensors V2. They are made by centralite with model number 3326-L. I have a bunch of these and would like to integrate them into homeassistant. There is an existing thread related to these sensors here - https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1731.

redwoodimage, 2020-03-04:
Thanks it working now

closed 2020-03-04 by redwoodimage