dresden-elektronik/deconz-rest-plugin #2461: Add Support for TS0201 Temperature and Humidity Sensor with Display

kroimon, 2020-02-17:
Manufacturer: TuYa? BlitzWolf? (seems to be sold under different brands)
Model Identifier: TS0201
Bought from: https://banggood.app.link/pfaR2ccl93

[image]
[image]
[image]
[image]
[image]

Darkwin101, 2020-02-17:
#2442 is the same Request

kroimon, 2020-02-18:
Duplicate of #2442

closed 2020-02-18 by kroimon