dresden-elektronik/deconz-rest-plugin #2462: Osram smart plug+ not reacting on all WebUI commands

DerOetzi, 2020-02-17:
I have an Osram smart plug+ with strange behavior:

- ON/OFF via IKEA switch is working
- ON/OFF via Openhab HUE Binding not working
- ON/OFF via Phoscon not working
- Switch OFF all lights in Phoscon is working

I just updated my docker container to 2.05.73 but same behaviour with 2.05.71 before the update.

All other TRADFRI and GLEDOPTO lights are reacting fine to the command of Phoscon and Openhab.

vandalon, 2020-02-18:
I see the exact same behaviour with 2 of my osram plugs.

vandalon, 2020-02-18:
after shutting down my pi and removing power it all works again it seems.

vandalon, 2020-02-19:
Having said that, it seems the 2 osram plugs I have now have a life of their own. The switch on without anything triggering it.

joch, 2020-02-19:
I started having issues with some of my Osram plugs and a light strip after their firmware was updated. I recently moved from marthoc/deconz to using Hassio, so they were upgraded automatically as part of that process.

In VNC, I selected the light strip and pressed `0`, which seems to have made it respond again (I don’t know that pressing 0 actually does, but it has been suggested in the past). I’ll try that with a plug which is having issues as well.

acid115, 2020-02-22:
I dont have problems with my 4 osram smartplugs (V1.04.12) and my 4 ledvance smart plugs (v00103101). I can switch the plugs on and off through phoscon web app and all other systems, what use Rest api, tested 30min ago.
I am on 2.05.73 too, since yesterday :)

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix