dresden-elektronik/deconz-rest-plugin #2464: Develco Smart Relay features

espen4001, 2020-02-18:
Hi! I have a Develco Smart Relay which, in addition to perform on/off-control of connected load, has the ability to send power measurement data, i.e. consumed power, voltage level, phase voltage etc.

How can I get these features available in my Home Assistant setup? They are available in deCONZ, but the Phoscon-app show only a single switch with on/off feature, and in Home Assistant, this switch is the only entity available.

I did try using the native ZHA integration in Home Assistant, and when the relay was connected, all features was available in Home Assistant; power measurement data, switching on/on etc.

I am pretty new to this ZigBee-stuff, so if this is something that needs modifying the API from deCONZ, or if I can do this by myself by modifying the unit in deCONZ, please help pointing me in the right direction. I can post a more "official" feature request if this is needs modification in the core of deCONZ.

Ref technical manual from manufacturer: 

[https://www.develcoproducts.com/media/1793/smrzb-332-technical-manual-smart-relay-din-16a.pdf](url)

**2.2 Key features**
The ZigBee based Smart Relay is a router in the ZigBee Pro network.
Key features are:
-  Remote on/off control
-  Power Failure alarm
-  Accurate power consumption
-  Certified ZigBee Home Automation 1.2
-  ZigBee OTA cluster for firmware upgrades
-  ETSI compliant.
-  RoHS compliant according to the EU Directive 2002/95/EC.

SwoopX, 2020-02-18:
Hi, I can take car of this when I've some time to spare. Meanwhile, does the device create the ZHAconsumption and ZHApower sensors alright in deconz? I understand HA did create something for power measurement and would assume, the values are empty?

Btw, you won't see any more in Phoscon than the light or switch you currently see. Phoscon is a different software and closed source, so nothing we can do about it. What matters here is what the REST API offers.

espen4001, 2020-02-18:
Thank you! I can provide screenshots from Deconz, maybe you can tell me what you need to see? As far as I can see, all the features listed in the manual are available.

espen4001, 2020-02-18:
Regarding what you are asking for about the ZHA. Not sure if I understand what you are asking. The only thing created in HomeAssistant from Deconz, is the switch on/off. Nothing else. When using the native ZHA in HomeAssistant, there is alot more. Current consumption, consumption last period, on/off, voltage and power fault alarm.

But please feel free to explain, as I really want to learn more about how zigbee works!

SwoopX, 2020-02-18:
It think I got everything I need for the moment regarding the screenshots. Maybe one of the node info panel, as Develco devices showed quite stubborn in some cases.

Bringing in support should be quite easy as I already implemented most of what's painful.

espen4001, 2020-02-18:
[image]

Here is a screenshot of the node-info view. I really appreciate your help on this!

espen4001, 2020-02-19:
I will get you some screenshots tomorrow!
I see you made a commit with initial support, I can update and see if you got the units correct.

espen4001, 2020-02-20:
Electrical measurement cluster - Relay On:
[image]

Electrical measurement cluster - Relay off:
[image]

Simple metering cluster - Relay on:
[image]

Simple metering cluster - Relay off:
[image]

Basic cluster - Relay off:
[image]

SwoopX, 2020-02-20:
Thanks. Looks like only the voltage required an update.

espen4001, 2020-03-03:
Finally got my docker updated to 2.05.74. But still nothing more than the «light» entity in home assistant. Shouldn’t I expect to see more entities? Or did I misunderstand it all? Am I mistaken by thinking that deConz talks directly to home assistant? That the phoscon app is only a more visual interpretation of the functionality of the Deconz?

SwoopX, 2020-03-03:
> That the phoscon app is only a more visual interpretation of the functionality of the Deconz?

Pretty much correct.

You may need to repair the device to have the sensors available. I'd recommend to go for the sensor search then. You should see something greenish upon success.

espen4001, 2020-03-04:
Thanks for the tip @SwoopX! A new search for lights in phoscon did the trick. Now I have two more entities in Home Assistant; Total consumption (Current summation delivered) and the current demand (Instantaneous demand). Just what I was looking for!
I also see that the device is able to measure temperature. Is it too much to ask for this value to be available also? Am I correct if this is cluster 0002, ID 0x0000?
Screenshot of the cluster info of the Device Temperature configuration:
[image]

SwoopX, 2020-03-04:
@espen4001 Glad it worked out, great. Regarding the device temperature, this is a rather extensive change I'm afraid. As I see it, changes on the API and sensor creation routines are required for that "compareable simple request". However, I'd vouche for that.

Maybe we should put that and the alarms cluster as well for discussion.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

jd1900, 2020-09-11:
> Hi! I have a Develco Smart Relay which, in addition to perform on/off-control of connected load, has the ability to send power measurement data, i.e. consumed power, voltage level, phase voltage etc.

Where did you buy it?
I'm trying to buy it but I cannot find it...

espen4001, 2020-09-12:
> > Hi! I have a Develco Smart Relay which, in addition to perform on/off-control of connected load, has the ability to send power measurement data, i.e. consumed power, voltage level, phase voltage etc.
> 
> Where did you buy it?
> I'm trying to buy it but I cannot find it...

https://www.elektroimportoren.no/develco-smartrele-16a-din/4514734/Product.html?Event=brandlist

jh83, 2020-09-15:
> > Hi! I have a Develco Smart Relay which, in addition to perform on/off-control of connected load, has the ability to send power measurement data, i.e. consumed power, voltage level, phase voltage etc.
> 
> Where did you buy it?
> I'm trying to buy it but I cannot find it...

Bought 4 of them from here: https://en.cozify.fi/collections/devices/products/develco-smart-relay-16a-din

Delivered to Sweden in less than a week