dresden-elektronik/deconz-rest-plugin #2465: Turning on one light in the group turns all lights in the group

andreypolyak, 2020-02-18:
Hi all,
I have the strange behavior of my lights in deCONZ (I'm using the latest version with Conee2). When I turn on one light in the group (no difference how: from Phoscon, GUI or Home Assistant) all lights in the group are turned on. I made a short GIF to show this behavior:
[image]

In this group, I have three IKEA E14 white temperature bulbs and Heiman smart plug. I have the same behavior in other groups:
- 3xIKEA E27 color bulbs
- 2xIKEA E27 white temperature bulbs and IKEA LED driver

Does anyone know why it happens and how to fix it?

Thank you!

maikkoe, 2020-02-19:
Hi,
flip your phone in landscape and check the state of the slider. You can select single or all. I hope this will help you. 
[image]

andreypolyak, 2020-02-19:
Hi @maikkoe,
I made this video on the phone just because it was easier to demo it. In fact, all lights in the group are turning on when this slider is in both positions or when I'm turning only one light in Home Assistant or when I'm turning only one light using deCONZ GUI.
But anyway, thanks for suggesting!

andreypolyak, 2020-02-20:
Guys, I found the reason. I forgot that I have Circadian Light component for these groups in Home Assistant. So when one light was turned on, this component set the color/temperature for all lights in the group and therefore turned them all.

Sorry for this, next time I’ll double check everything. I’m closing this issue.

closed 2020-02-20 by andreypolyak