dresden-elektronik/deconz-rest-plugin #2466: No NodeAdded event for gateway node?

ebaauw, 2020-02-18:
I'm adding support for exposing a RaspBee, ConBee, ConBee II, or XBee as a `/lights` resource, like the Trådfri repeater.

I've got it working, see the resources below, but the resource for the gateway itself doesn't get re-created after restarting deCONZ.  As far as I can tell, it's because the REST API plugin re-creates the `/lights` resources on receiving the `deCONZ::NodeEvent::NodeAdded` event from the deCONZ core.  I now re-create the resource when the gateway node is selected, but that's a bit of a dirty workaround.

@manup, Is it possible for the deCONZ core to issue a `NodeAdded` event on startup, when it creates the node for the gateway?  Alternatively, is there a way for the REST API plugin to get the node for the gateway, other than through a `deCONZ::NodeEvent`, so I could simply call `addLightNode()` during initialisation of the REST API plugin.

Slightly related: Where did the documentation for the C++ API for plugins move to?  I bookmarked https://www.dresden-elektronik.de/fileadmin/Downloads/Dokumente/Produkte/6_Software/deconz-cpp-doc/index.html, but that link is dead.

```json
{
  "etag": "a0647a3cd0f449c9ab33bf3bf5fbb32b",
  "hascolor": false,
  "manufacturername": "dresden elektronik",
  "modelid": "RaspBee",
  "name": "RaspBee",
  "state": {
    "reachable": true
  },
  "swversion": "0x26330500",
  "type": "Configuration tool",
  "uniqueid": "00:21:2e:ff:ff:01:04:1e-01"
}
```
```json
{
  "etag": "f108d1aeed298763d8b107f1a55ce2d8",
  "hascolor": false,
  "manufacturername": "dresden elektronik",
  "modelid": "ConBee II",
  "name": "ConBee II",
  "state": {
    "reachable": true
  },
  "swversion": "0x264a0700",
  "type": "Configuration tool",
  "uniqueid": "00:21:2e:ff:ff:03:d4:a7-01"
}
```
```json
{
  "etag": "5d9b7fbf2dede4ae98f2a0c1512167c7",
  "hascolor": false,
  "manufacturername": "Digi",
  "modelid": "XBee",
  "name": "XBee",
  "state": {
    "reachable": true
  },
  "swversion": null,
  "type": "Level control switch",
  "uniqueid": "00:13:a2:00:41:83:1c:ae-e6"
}
```
I fixed and expanded the _Basic_ cluster for the gateway:
[image]

Smanar, 2020-02-19:
Hello,
I don't see where you have make change to have Model id/Manufacture, and ect ...
It's just in de_web_plugin.cpp ?

ebaauw, 2020-02-19:
It’s in an earlier commit: https://github.com/ebaauw/deconz-rest-plugin/commit/42011c6b14ab10996d0db406a0dea7b4f310658f.

Smanar, 2020-02-19:
Ha ok, lol, I have missed the previous commit, it's ok now.

ebaauw, 2020-02-19:
I'm happy to add it; I just wasn't sure which code to use (and too lazy to investigate).  You might want to check that I didn't break the Legrand pairing.  The _ZCL Version_ value was written without casting it to the correct data type, causing the GUI to grey out the attribute.  For good measure, I also cast the 0xF000 value, in case deCONZ runs on a system with `int` size other than 32 bits.

manup, 2020-02-19:
> @manup, Is it possible for the deCONZ core to issue a NodeAdded event on startup, when it creates the node for the gateway? Alternatively, is there a way for the REST API plugin to get the node for [...]

The easiest way is to use the ApsController::getNode() method, that should also return the coordinator note (likely at index 0).

```c++
    /*! Iterates through the internal node cache.
        \code {cpp}
        int i = 0;
        const deCONZ::Node *node;
        deCONZ::ApsController *ctrl = deCONZ::ApsController::instance();

        while (ctrl->getNode(i, &node) == 0)
        {
          // do something with *node
          i++;
        }
        \endcode

        The node with the index 0 is always the own node.

        \param index a index >= 0
        \param node a pointer to a user provides deCONZ::Node pointer
        \retval 0 if \p node points to a valid deCONZ::Node
        \retval -1 if no node is available for \p index
     */
    virtual int getNode(int index, const Node **node) = 0;
```

>Slightly related: Where did the documentation for the C++ API for plugins move to? I bookmarked https://www.dresden-elektronik.de/fileadmin/Downloads/Dokumente/Produkte/6_Software/deconz-cpp-doc/index.html, but that link is dead.

A good catch totally forgot that, I'll see to relocate that to phoscon.de.

ebaauw, 2020-02-19:
> The easiest way is to use the ApsController::getNode() method, that should also return the coordinator note (likely at index 0).

Thanks!  According to the comments in `/usr/include/deconz/aps_controller.h`:
> The node with the index 0 is always the own node.

So this should also work when configured as router ;-).

ebaauw, 2020-02-19:
Created PR #2472.

closed 2020-02-19 by ebaauw

Smanar, 2020-02-19:
@ebaauw 
Thx, I have tested the code, it's still working, so no problem.