dresden-elektronik/deconz-rest-plugin #2467: deCONZ app but ConBee II dongle on remote host

rotragit, 2020-02-19:
Hi all, I have a question:
I have the ConBee II dongle connected to a PC running UBUNTU server. The PC has not any monitor and X11 is not installed. deCONZ is installed by docker image and I prefer not to change the type of installation. That way I can access using the web GUI but I don't have the deCONZ desktop app.
There is any way to install the deCONZ app on a different PC and tell the app to connect to the remote deCONZ?

I've tried also usbip(d), but it doesn't run if the dongle is in use by the docker container.

Finally, the network configuration is saved on the ConBee dongle or is saved in the deCONZ module?  I mean, if I stop the container and make the usbip tunnel for a while, I should see the nodes of my network?

Thank you!

Smanar, 2020-02-19:
All is saved on machine host, not in the dongle.

And no, I don't think it's possible having the GUI on a computer, and the headless app on another, because the GUI is the application too.

rotragit, 2020-02-21:
Thank you, I've installed a desktop, and I was able to launch deCONZ from the docker image exporting DISPLAY and adding /tmp/.X11-unix as a volume to the container. The app shows up ok and I see the option to connect to the ConBee II dongle. But when I click on the "Connect" buttom nothing happen.
I have no specific errors on the logs. There is any way to debug / increase the verbouse of the log?

The PC I use for deCONZ, is also used to monitor cameras, home assistant, 433Mhz sensors, pihole, mqtt server and a lot of other stuff. It's really a problem to install deCONZ not in a docker container because of lib dependencies with the other applications.

Smanar, 2020-02-21:
Yep you can add some debug information if you run it with command line "--dbg-info=1 --dbg-error=1 --dbg-aps=1 --dbg-zdp=1"

For docker there is a link just for that https://phoscon.de/en/conbee/install#docker
But I m not sure to understand if you use a docker or not ?

You have a combo box with conbee and raspbee above the "connect button" ?
I think phoscon have "not connected" as firmware ?

rotragit, 2020-02-21:
Yes, I'm running in docker and yes, I see the combo but is selected conbee and I cannot drop-down the combo. OK, so the problem could be the desktop manager I'm using (firefox has the same problem if it's started in a desktop-manager but there is not a windows-manager above it, I know that because I have a lot of kiosk-mode terminals at work).

Firmware is loaded:
17:18:50:502 GW firmware version: 0x264a0700
17:18:50:502 GW firmware version is up to date: 0x264a0700

and I'm able to connect to the web gui.

OK, I'll investigate :-)

Just for your reference, I started deCONZ with docker-compose with the following yaml:

`version: "2"
services:
  deconz:
    image: marthoc/deconz
    container_name: deconz
    network_mode: host
    restart: always
    volumes:
      - /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ
      - /tmp/.X11-unix:/tmp/.X11-unix
    devices:
      - /dev/ttyACM0
    environment:
      - DECONZ_WEB_PORT=8080
      - DECONZ_WS_PORT=8443
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DISPLAY
`
To start the app I have first to enable X authentication with:

`$ xhost +local:docker`

then:

`$ docker exec -e -it $(containerID) /usr/bin/deCONZ`

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix