dresden-elektronik/deconz-rest-plugin #2469: How to access Rest API- No 'Gageway unlock'

Liese2007, 2020-02-19:
Hi, I try to work with REST API, but the button mentiond in documentation is missing... Imho.
Please check pictures:
[image]
[image]
 Regards
Jörg

ademobile, 2020-02-19:
I believe you want 'App verbinden'. Sorry I don't read/speak German, however that's the button I use, if I have understood you correctly.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix