dresden-elektronik/deconz-rest-plugin #2470: issue after re-adding hue dimmers with upgraded firmware

andreibanaru, 2020-02-19:
I have upgraded my Hue Dimmers to 6.1.1.28573 and re-added them to deconz.
Then mapped my nodes to them but I can't control the lights anymore.

[image]

Whenever I push the button I see this in the logs:
19:43:48:270 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 004
19:43:48:446 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 004
19:43:53:958 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 001
19:43:54:141 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 001
19:43:55:467 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 004
19:43:55:607 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 004
19:43:56:489 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 003
19:43:56:639 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 003
19:43:57:190 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 002
19:43:57:340 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 002

Using the Switch Editor works but that's not desired as I want some independence from the deconz gateway.

manup, 2020-02-19:
Seems like the connection is broken (behind the scenes group cast), have you tried removing and adding a light to the switch again. "Edit > Manage Lights"

andreibanaru, 2020-02-19:
I had to "reboot" the switch. Keep all 4 buttons pressed at the same time for a couple of seconds.

closed 2020-02-19 by andreibanaru

jimmy927, 2020-04-25:
> I had to "reboot" the switch. Keep all 4 buttons pressed at the same time for a couple of seconds.

I had the exactly same issue, came across your post.
Keep all 4 buttons pressed at the same time for a couple of seconds helped!

But this is a new issue with deconz, previously this was not needed.

Information on my RWL021:

Version 5.45.1.17846

i guess thats rather old, but not sure.