dresden-elektronik/deconz-rest-plugin #2473: `/lights` resources not available after starting deCONZ

ebaauw, 2020-02-20:
Seems v2.05.73 re-introduced an old bug: after starting deCONZ, the REST API plugin doesn’t expose any `/lights` resources.  It takes a minute or so, before they’re available.  I also noticed the GUI stays on the device selection screen for quite some time, and then displays an empty main screen.  The nodes only appear after a while.

If homebridge is started before all `/lights` resources are available, HomeKit removes the lights.  If homebridge is restarted, the lights re-appear in HomeKit, but as new lights (losing their HomeKit name and assignments to HomeKit rooms, groups, scenes, and automations).  Create a “lightlink” resourcelink, to make homebridge-hue delay homebridge startup until all lights in the resourcelink are available, see https://github.com/ebaauw/homebridge-hue/issues/471.

SwoopX, 2020-02-20:
I experienced this as well just recently after editing my database and got a little hectic... But yes, it all came back.

ebaauw, 2020-08-15:
I hope adding the Backlog label will shut up the bloody bot.

silent-303, 2020-10-24:
Thanks for your attention. It would be very nice if this gets fixed.

Schosch83, 2020-10-29:
I have the same problem on deconz 2.05.86 beta or 2.05.84 stable.

simplenotezy, 2020-11-01:
Got the same issue. Spend an hour or so trying out deCONZ. After a reboot, all work was lost. Would be nice to have fixed.

Lucleonhart, 2020-11-06:
Same issue with brand new Homebridge + deconz installation. -_-

gruenberg, 2020-11-10:
I can suggest this workaround, in my setup it works fine:

sudo systemctl stop deconz-homebridge.service
cd /home/pi/
sudo chown -R pi:pi .homebridge

--> edit file /etc/rc.local and add these two lines before exit

sleep 180
sudo systemctl restart deconz-homebridge.service &

--> hue lights should re-appear in 3.5 minutes in Apple Home App after restart/boot of phoscon.local

Schosch83, 2020-11-10:
@gruenberg Thank you for you're workaround. 
For me was not working! But I found a better way, from @ebaauw.

in Terminal you write:
ph createuser

(now you get a user for ph)

ph lightlist

(now he added all lights to a waiting list)

ph lightlist -v
(he show the lights from deconz)

And now Homebridge waiting until all the lights on the ph lightest are there.
Thats work for me and hopefully for others.

simplenotezy, 2020-11-11:
@Schosch83 can you link to the original message? Thanks

Schosch83, 2020-11-11:
@simplenotezy 
I found ist here:
https://github.com/ebaauw/homebridge-hue/issues/246
Hopefully I could help.

simplenotezy, 2020-11-23:
@Schosch83 sorry, don't see any instructions there :/

Schosch83, 2020-11-23:
@simplenotezy Sorry that you not see the workaround. But is working or?

Mimiix, 2020-12-16:
@manup Fix.

simplenotezy, 2020-12-23:
@Schosch83 hehe, I don't know if its working or not, untill you send me the instructions on how to apply the fix :P

Anyhow, I have hoped that dresden-elektronik would fix this issue sooner, but my patience is up. I'll have to switch back to using my Phillips Hue Bridge.

I have now lost all my lights, scenes, motion sensor setups etc, for all 5 rooms more than 4 times now. It takes me around 30 minutes each time to set it up again in HomeKit.

Thanks for an otherwise great product, I really enjoyed using it!

gruenberg, 2020-12-23:
> @Schosch83 hehe, I don't know if its working or not, untill you send me the instructions on how to apply the fix :P
>
> Anyhow, I have hoped that dresden-elektronik would fix this issue sooner, but my [...]

just read what @Schosch83 commented on 10 Nov, it contains complete instructions and they do work.

Schosch83, 2020-12-23:
@simplenotezy This is of course annoying. What do you get if you enter: ph lightlist -v in the terminal?

manup, 2020-12-30:
Can anybody confirm it is working now as expected with [v2.9.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.9.0-beta)?

ebaauw, 2020-12-30:
It is.  The `/lights` are returned right after startup, even before the GUI is displaying the map.  Checked PiOS Stretch with RaspBee, PiOS Buster with ConBee II, and PiOS Buster with RaspBee II.

As I mentioned on Discord I had some segmentation faults starting v2.9.0-beta on Raspbian Stretch using RaspBee 26390500, but it seems fine now.

simplenotezy, 2020-12-31:
@ebaauw does that mean it is working now?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

simplenotezy, 2021-01-06:
@Schosch83 I tried your suggestion, but upon running first command on homebridge I get:

```
pi@homebridge:~ $ ph createuser
ph: request 1: GET /config
ph: warning: 127.0.0.1:80: cannot connect: ECONNREFUSED
ph: error: 127.0.0.1:80: cannot connect: ECONNREFUSED
ph: fatal: localhost: not a Hue bridge nor deCONZ gateway
```

simplenotezy, 2021-01-06:
> It is. The `/lights` are returned right after startup, even before the GUI is displaying the map. Checked PiOS Stretch with RaspBee, PiOS Buster with ConBee II, and PiOS Buster with RaspBee II.
>
> As [...]

@ebaauw Still not sure whether or not I should try to implement the workaround, or whether or not your comment here, says that it is now working? 😊

simplenotezy, 2021-01-06:
My current workaround is to create some scenes and motion detection rules in Phoscon app, then I guess I won't have to remake them everytime. I would still have to rearrange my rooms every time though.

I have updated all plugins to latest version, as well as node and homebridge, and will report back here, if the issue persists.

Feedback for phoscon: Would be nice if you could create scene rules that only would be applied if somebody was at home / if the lights was already on at some brightness level. Then you could automate to dim the lights at night, but only if they are already turned on at say 80% or if I am home (so they won't turn on automatically when I am not at home)

ebaauw, 2021-01-06:
> Still not sure whether or not I should try to implement the workaround, or whether or not your comment here, says that it is now working?

What don’t you understand?

> > Can anybody confirm it is working now as expected with v2.9.0-beta?

> It is.

So the workaround is no longer needed for v2.9.0-beta.

> I tried your suggestion, but upon running first command on homebridge I get

Specify the ip address and port deCONZ is using, see `ph -h`.

closed 2022-10-15 by ebaauw