dresden-elektronik/deconz-rest-plugin #2474: LifeContol MCLH-03 smart plug, each 5-10 mins it switches off the relay for just a second and then switches it on again.

macros0, 2020-02-20:
> @bastshoes I have updated Deconz and re-added the plug into Deconz. Energy power and consumptions have added but the plug itself become a disaster. Now each 5-10 mins it switches off the relay for just a second and then switches it on again. At the same time there is no any event in HA log regarding it, HA believes that nothing happens. Thus now I can't use the plug because it so regularly restarts a plugged device.

What to do, bought a plug in too have all these problems?

_Originally posted by @macros0 in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1573#issuecomment-589000202_

bastshoes, 2020-02-20:
Try to reduce report threshold for wattage, current and voltage. This can be done via GUI interface.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix