dresden-elektronik/deconz-rest-plugin #2475: Refactor logic setting `state.on` and `state.bri`

ebaauw, 2020-02-20:
Better open a separate issue for this, as it tales me forever to search through closed issues and find my old https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1111#issuecomment-456464872.

Still very relevant, see e.g. #1594.

ebaauw, 2020-03-08:
Because the lights have no associated `ResourceItem` for `state.effect`, it isn't reported in the web socket notification.

ebaauw, 2020-03-09:
Cross-referencing https://github.com/dresden-elektronik/deconz-rest-plugin/issues/973.

ebaauw, 2020-04-05:
Next iteration: the above commits add `state.open`, `state.lift`, and `state.tilt` to _Window Covering_ devices:
- `state.lift` is best understood as "percentage closed".  It reflects _Current Position Tilt Percentage_ from 0% (fully open) to 100% (fully closed).  Note that some devices (like Xiaomi) reverse the values; the REST API plugin handles this, so `state.lift` works the same for all devices;
- `state.open` is derived from `state.lift`: it is true while `state.lift` < 100%.
- `state.tilt` reflects _Current Position Tilt Percentage_, from 0% to 100%.
- Setting `state.lift` to a value between 0 and 100 results in a _Goto Lift Percentage_ command.
- Setting `state.lift` to `"stop"` results in a _Stop_ command.
- Setting `state.open` to true results in an _Up/Open_ command.  Setting it to false will issue a _Down/Close_ command.
- Setting `state.tilt`to a value between 0 and 100 results in a _Goto Tilt Percentage_ command.

```
$ ph get /lights/327
{
  "etag": "1450b426c7a26529ac316db62e01b8a5",
  "hascolor": false,
  "lastseen": null,
  "manufacturername": "LUMI",
  "modelid": "lumi.curtain",
  "name": "Bedroom Curtains",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": true,
    "open": false,
    "reachable": true
  },
  "swversion": "04-13-2017",
  "type": "Window covering device",
  "uniqueid": "00:15:8d:00:02:83:4c:6c-01"
}
```

Note that not all _Window Covering_ devices support all these commands.  Notably the Xiaomi Curtain Controller B1 (that uses the _Analog Output_ cluster), only supports (the equivalent) of _Goto Lift Percentage_.

The `state.on`, `state.bri`, and `state.sat` attributes are still supported, and work as before.  However, when setting these, the REST API responds with the new attributes:
```
$ ph put -v /lights/327/state '{"on": false}'
{
  "open": true
}
$ ph put -v /lights/327/state '{"bri": 254}'
{
  "lift": 100
}
```

Both old attributes are included in the websocket notifications (note: I'm running with `websocketnotifyall` set to false):
```
Apr 05 23:13:25 pi2 dc_eventlog[860]: /lights/327/state: {"alert":null,"bri":182,"lift":72,"on":true,"open":true,"reachable":true}
Apr 05 23:13:30 pi2 dc_eventlog[860]: /lights/327/state: {"bri":0,"lift":0,"on":false}
Apr 05 23:14:15 pi2 dc_eventlog[860]: /lights/327/state: {"bri":33,"lift":13,"on":true}
Apr 05 23:14:20 pi2 dc_eventlog[860]: /lights/327/state: {"bri":254,"lift":100,"open":false}
```

gmitch64, 2020-05-28:
Just to throw something in here...

> The fan is exposed though an added speed attribute. I'm still not sure whether the Fan Control cluster should better be exposed as a separate (for now) /lights resource.

When looking physically at the device (and talking about the Hamton Bay one you've just fixed - thank you), you will ALWAYS have the fan. There may be an attached light unit, or there may no - it's optional whether you install it or not. Now, I don't know if the the remote can actually tell you if the light kit has been installed or not.

I'd think in this case, the "whole fan", should be presented up in a similar method to the multi sensors - with a sub device for the fan, and a sub device for the lights - like a multi sensor has a device for temp, a device for pressure, etc.

G

ebaauw, 2020-05-28:
> When looking physically at the device

I appreciate that you do that, but it's irrelevant for how a device is exposed through the REST API.  The resource structure follows what the device looks like from a Zigbee perspective: the exposed endpoints and clusters.  See my explanation in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/932#issuecomment-635618022.

ebaauw, 2020-06-01:
OK, PR #2553 caused some more effects that I had anticipated.
- Specifying invalid values, that were previously silently ignored, or coerced to valid values, now result in error messages (e.g. setting `ct` outside the `ctmin` to `ctmax` range, see #2798; or setting `colormode`, see #2840);
- Specifying inappropriate commands, that accidentally worked, now result in error messages and no longer work (e.g. `ct` for a _Color light_, that doesn't support _Color temperature_, see #2798).

I also found some other weird things:
- Allowing both integer and string (`"stop"`) values to `state.lift` is highly inconsistent, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-636520877.
- Some lights incorrectly report `ctmax` as 0, see #2853;
- According to the ZCL spec, the maximum value for _Color Temperature_, _Color Temperature MIn_ and _Color Temperature Max_ is 65279 (0xFEFF), same as for _Current X_ and _Current Y_, see #2856.

For my next attempt, I propose the following:
- Only report an error for violations of the data type (e.g. setting setting `ct` to `true`, or setting `ct` to an integer value < 0 or > 0xFFFF).  Coerce the value if of the correct type, but invalid (7 is a Hue light; 10 a Trådfri):
   ```
  $ ph put -v /lights/7/state '{"on": true, "ct": true}'
  ph put: fatal: invalid value, true, for parameter, ct.
  $ ph put -v /lights/7/state '{"on": true, "ct": -1}'
  ph put: fatal: invalid value, -1, for parameter, ct
  $ ph put -v /lights/7/state '{"on": true, "ct": 0}'
  {
    "on": true,
    "ct": 153
  }
  $ ph put -v /lights/7/state '{"on": true, "ct": 10000}'
  {
    "on": true,
    "ct": 500
  }
  $ ph put -v /lights/10/state '{"on": true, "ct": 0}'
  {
    "on": true,
    "ct": 250
  }
  $ ph put -v /lights/10/state '{"on": true, "ct": 10000}'
  {
    "on": true,
    "ct": 454
  }
  $ ph put -v /lights/7/state '{"on": true, "xy": [2,1]}'
  ph put: fatal: invalid value, [2,1], for parameter, xy
  $ ph put -v /lights/7/state '{"on": true, "xy": [1,1]}'
  {
    "on": true,
    "xy": [
      0.9961,
      0.9961
    ]
  }
  ```
- Return HTTP status 200 on API errors (just like the Hue bridge).  This means that invalid keys or values in the body are ignored, but valid ones are still executed (with an experimental version of `ph` to handle this):
  ```
  $ ph put -v /lights/7/state '{"on": false, "ct": 1000, "xy":[1,2]}'
  {
    "warnings": [
      "invalid value, [1,2], for parameter, xy"
    ],
    "ct": 500,
    "on": false
  }
  $ ph put -v /lights/7/state '{"on": false, "ct": 1000, "xy":[1,2], "key": "invalid"}'
  {
    "warnings": [
      "parameter, key, not available",
      "invalid value, [1,2], for parameter, xy",
      "parameter, ct, is not modifiable. Device is set to off."
    ],
    "on": false
  }
  ```
- Add a sanity check that `ctmax` > `ctmin`.  Change default `ctmax` to 65279.

Smanar, 2020-06-01:
For window covering, I m agree, and for retro-compatibiliy there is still {"bri_inc": 0}

closed 2024-09-04 by ebaauw