dresden-elektronik/deconz-rest-plugin #2476: Device Request LUPUS 12050 ZigBee

ricosander, 2020-02-20:
The device is recognized, but without measuring information on deconz adapter in iobroker.

[image]

deconz screenshots:
[image]
[image]
[image]

SwoopX, 2020-02-20:
Please issue a new device support request according to the wiki.

ricosander, 2020-02-21:
I tried with this issue. What can I do better?

SwoopX, 2020-02-21:
We'd require at lease a screenshot from the basic and simple metering cluster (for simple metering, there's the remaining attributes missing) with the attributes being read.

ricosander, 2020-02-21:
Basic READ:[image]
[image]

Simple Metering READ:
[image]
[image]

SwoopX, 2020-02-21:
Sorry, but you forgot to read the attributes before. All values are default and therefore not usable.

ricosander, 2020-02-26:
I changed the basic read picture in the post above.

SwoopX, 2020-02-26:
Thanks, I'll follow up the next couple of days. Btw, does the "instantaneous demand" attribute show any reasonable values if you apply a load to the plug?

SwoopX, 2020-02-27:
I checked the code and support is already implemented. You should definitively update your deconz version since it is rather old. After update, you may also restart iobroker.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix