dresden-elektronik/deconz-rest-plugin #2477: Conbee 2 bootloop on Raspberry PI 4 - 4GB

LukasA83, 2020-02-20:
I'm currently migrating my hosting environment onto a Raspberry PI 4b w. 4GB with a docker setup. During the migration I ran into the issue that Conbee 2 is on a bootloop:

```
[Fri Feb 21 00:50:28 2020] usb 1-1.4: SerialNumber: DE2120616
[Fri Feb 21 00:50:28 2020] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[Fri Feb 21 00:50:51 2020] usb 1-1.4: USB disconnect, device number 46
[Fri Feb 21 00:50:51 2020] usb 1-1.4: new full-speed USB device number 47 using xhci_hcd
[Fri Feb 21 00:50:51 2020] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Fri Feb 21 00:50:51 2020] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Feb 21 00:50:51 2020] usb 1-1.4: Product: ConBee II
[Fri Feb 21 00:50:51 2020] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Fri Feb 21 00:50:51 2020] usb 1-1.4: SerialNumber: DE2120616
[Fri Feb 21 00:50:51 2020] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```
System:
Raspberry PI4b w. 4GB
Linux ubuntu 5.3.0-1018-raspi2 #20-Ubuntu SMP Mon Feb 3 19:45:46 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
No other USB connected, WIFI, bluetooth disabled
Tried USB 2.0 and USB 3.0 ports w. & w/o extension cable

Conbee 2 works normally with a windows machine. Any idea what to try?

elafargue, 2020-02-21:
On my Pi 4, the Conbee 2 just crashes after a few minutes and takes the USB bus down with it, which looks like a variant of your problem. Do you know what firmware your Conbee 2 is running? it looks like the current version has serious issues on the Pi 4...

LukasA83, 2020-02-21:
Conbee 2 Firmware is latest, 264A0700. On my Pi it is not related to deconz. Even if I stop / remove the container, I still the bootloop in dmesg...

LukasA83, 2020-02-21:
Tried it also with Raspbian OS, latest updates, same behaviour. Any idea?

SwoopX, 2020-02-21:
Are you running the latest deconz version? Just read manup announced version .74 to arrive soon with further enhancements.

LukasA83, 2020-02-21:
Docker image is latest available .73. Not sure it directly related to deconz software, as even with the docker container stopped, dmesg indicates the sticks connects / disconnects every 5-60 seconds.

LukasA83, 2020-03-01:
Works with .74

closed 2020-03-01 by LukasA83

dvbit, 2021-01-03:
please reopen.
Issue is not solved

Mimiix, 2021-01-03:
@dvbit quit replying on old issues. I've already told you what to do.