dresden-elektronik/deconz-rest-plugin #2478: all devices offline due to powerloss

r2ixuz, 2020-02-21:
My home server crashed due to a powerloss. Deconz cannot reconnect to a single node. I had no Backup, so I needed to rejoin every node again. All worked besindes my Philips Hue White Ambiance E14 LED lamp/bulb.

I purchased the Hue lamp without any Philips remote and joined it directly into my Zigbee deconz network.

But now how can I rejoin it? I never had a Hue Bridge or remote control. :(

ebaauw, 2020-02-21:
Afaik the Hue lights can only be factory reset thru touchlink.  You might try the old web app, or any wireless switch that supports touchlink reset.  Make sure to powercycle the light first, as it only responds to touchlink shortly after powered on.

r2ixuz, 2020-02-21:
So you say I need to buy another piece of hardware for around 20 EUR to reset the lamp? This is ridiculous. I bought this lamp Nov. 2019.

And all that happend because I didn't know that every device won't work anymore after a powerloss. Would a backup via Phoscon app helped me out of this situation?

Smanar, 2020-02-21:
Like he said try first the touchlink feature in the old webapp before buying a remote, and remeber for this feature the bulb need to be realy close the conbee (around 50cm)

r2ixuz, 2020-02-21:
Ok walked with my lamp 10 cm close to the conbee stick. I powered the lamp on and clicked scan devices (Touchlink) via the old web app but nothing was found. :-(

Does Touchlink need UPNP? I needed to disable this in my Docker config.

r2ixuz, 2020-02-21:
misclicked...

Smanar, 2020-02-21:
None ? I can detect an ikea bulb at 3/4m ....
No, it's a zigbee feature, but are you sure your conbee is working ATM ? You have a channel and a network id ?

r2ixuz, 2020-02-21:
Yes, it says my gateway is on channel 15 and has id 0xa18c

r2ixuz, 2020-02-21:
Do I need to remove the lamp from the Phoscon app? All other device just worked again after repairing.

Smanar, 2020-02-22:
So all is working now, except philips bulb ?
On my side I have tried the touchlink feature with bulb already in network.

r2ixuz, 2020-02-22:
No matter what I tried the bulb did not reset. I even tried it with the IKEA Tradfyr rollo remote control...

So I ended up with buying this Philips dimmer remote and it worked...

closed 2020-02-22 by r2ixuz