dresden-elektronik/deconz-rest-plugin #2480: Tuyatec TSxxx switch series isn't drive

deennoo, 2020-02-22:
Hi there.

I have some tuyatec Switch (1.2.3 gang, L Only / L+N)

They are zigbee3 compatible, i can add them to my raspbee and my deconz.

I made some test with 2 of them : 

1 Gang line only : i can have some information return (on/off) but i can't drive them
2 Gang line only :  i can have some information return (on/off) but i can't drive them only for one gang, the second isn't shown.

Phoscon PWA give : 

Vendor_TYZB01_xfpdrwvc
ModelTS0011

What did you need for more help ? i can read all the cluster on GUI, but i don't know how to export them

Smanar, 2020-02-24:
Hello, someone have already started to include tuyatec, but I haven't see this device yet.
Usefull screenshot to make are here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

BTW, before making integration, better to make it working in deconz. Have you try to act on the device using deconz (cluster 0006 > on and off ? You have an error message or it just not working ?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix