dresden-elektronik/deconz-rest-plugin #2481: conbee II stopped working out of the blue

veloc1ty, 2020-02-22:
My conbee 2 was working flawlessly for 2 month, yesterday it just stopped out of the blue without changing anything and I am not able to revive it.

I noticed that I had no sensor readings since 4 hours and restarted the machine, since then, all I see is:

`16:19:37:751 Device protocol version: 0x010B
16:19:37:810 Current channel 15
16:19:37:831 CTRL ANT_CTRL 0x03
16:19:37:862 NET ZDO start network status 0xC4
16:19:42:615 try to reconnect to network try=7
16:19:42:631 Device protocol version: 0x010B
16:19:42:699 Current channel 15
16:19:42:718 CTRL ANT_CTRL 0x03
16:19:42:747 Device protocol version: 0x010B
16:19:42:798 Current channel 15
16:19:42:816 CTRL ANT_CTRL 0x03
16:19:42:844 NET ZDO start network status 0xC4`

It shows "firmware not connected" in the deconz web app.

Troubleshooting steps taken:
- Reboot, unplug, reinstall, use different USB port, re-flash firmware with GCFFlasher from windows and ubuntu. Tried to reset it and re-add the lights. No-go.

When I plug it in and run the deconz app, I can see the network, but I cannot do anything in there.
I am running the home assistant deconz integration. Also tried restoring a snapshot of that from like 2 weeks ago - didn't do anything.

It seems to be completely unable to add or communicate with my devices (Osram, Hue, Xiaomi sensors, nothing exotic).

The lights show up in the console, but I am not able to control them either from homeassistant or somewhere else. I can see them in the deconz app, but there is no mesh, they just show up as individual nodes.

`16:23:30:960 started websocket server at port 8081
16:23:30:962 found node plugin: libde_rest_plugin.so - REST API Plugin
16:23:30:964 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
[16:23:31] INFO: Starting Nginx...
[16:23:31] INFO: Running Hass.io discovery task...
2020/02/22 16:23:31 [notice] 510#510: using the "epoll" event method
2020/02/22 16:23:31 [notice] 510#510: nginx/1.14.0 (Ubuntu)
2020/02/22 16:23:31 [notice] 510#510: OS: Linux 4.19.104
2020/02/22 16:23:31 [notice] 510#510: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/02/22 16:23:31 [notice] 510#510: start worker processes
2020/02/22 16:23:31 [notice] 510#510: start worker process 518
[16:23:31] INFO: Running the deCONZ OTA updater...
[16:23:31] INFO: Running the IKEA OTA updater...
[16:23:31] INFO: Running the LEDVANCE/OSRAM OTA updater...
[16:23:31] INFO: deCONZ is set up and running!
16:23:32:553 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
16:23:32:568 COM: /dev/ttyACM0 / serialno: DE2132713
16:23:32:568 COM: --dev: /dev/ttyACM0 (ConBee II)
16:23:32:672 Device firmware version 0x264A0700
16:23:32:680 unlocked max nodes: 200
16:23:32:761 Device protocol version: 0x010B
16:23:32:769 new node - ext: 0x00212effff050589, nwk: 0x0000
16:23:32:833 SensorNode 2 set node 0x00158d000407fd4a
16:23:32:834 SensorNode 3 set node 0x00158d000407fd4a
16:23:32:834 SensorNode 4 set node 0x00158d000407fd4a
16:23:32:856 SensorNode 5 set node 0x00158d00044bf1c4
16:23:32:864 SensorNode 6 set node 0x00158d000418e7d3
16:23:32:864 SensorNode 7 set node 0x00158d000418e7d3
16:23:32:864 SensorNode 8 set node 0x00158d000418e7d3
16:23:32:871 SensorNode 9 set node 0x00158d00044bdb75
16:23:32:878 SensorNode 12 set node 0x00158d00044abd4c
16:23:32:878 SensorNode 10 set node 0x00158d00044abd4c
16:23:32:879 SensorNode 11 set node 0x00158d00044abd4c
16:23:32:885 SensorNode 13 set node 0x00158d0003d42449
16:23:32:886 SensorNode 14 set node 0x00158d0003d42449
16:23:32:886 SensorNode 15 set node 0x00158d0003d42449
16:23:32:892 SensorNode 16 set node 0x00158d00044d00ec
16:23:32:892 SensorNode 17 set node 0x00158d00044d00ec
16:23:32:899 SensorNode 18 set node 0x00158d0003f3775d
16:23:32:899 SensorNode 19 set node 0x00158d0003f3775d
16:23:32:905 SensorNode 20 set node 0x00158d00041f7572
16:23:32:912 don't close database yet, keep open for 900 seconds
16:23:32:913 LightNode 3: Flur added
16:23:32:920 don't close database yet, keep open for 900 seconds
16:23:32:920 LightNode 4: Tisch added
16:23:32:927 don't close database yet, keep open for 900 seconds
16:23:32:927 LightNode 5: Twiggi added
16:23:32:933 don't close database yet, keep open for 900 seconds
16:23:32:934 LightNode 6: Wagenfeld added
16:23:32:940 don't close database yet, keep open for 900 seconds
16:23:32:941 LightNode 7: Deckenlampe Küche added
16:23:32:947 don't close database yet, keep open for 900 seconds
16:23:32:947 LightNode 8: xxxxxx added
16:23:32:954 don't close database yet, keep open for 900 seconds
16:23:32:955 LightNode 9: xxxxx added
16:23:32:962 don't close database yet, keep open for 900 seconds
16:23:32:962 LightNode 10: xxxx added
16:23:32:969 don't close database yet, keep open for 900 seconds
16:23:32:969 LightNode 11: xxxxadded
16:23:32:976 don't close database yet, keep open for 900 seconds
16:23:32:976 LightNode 12: Drucker added
16:23:32:983 SensorNode 22 set node 0x001788010673805d
16:23:32:989 SensorNode 24 set node 0x00178801020e0be5
16:23:33:003 don't close database yet, keep open for 900 seconds
16:23:33:003 LightNode 13: Arbeitszimmer Tisch added
16:23:33:011 SensorNode 23 set node 0x00158d0004451e6c
16:23:33:018 SensorNode 25 set node 0x00158d000405cfb4
16:23:33:111 Current channel 15
16:23:33:131 CTRL ANT_CTRL 0x03
16:23:35:755 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
16:23:35:755 GW firmware version: 0x264a0700
16:23:35:755 GW firmware version is up to date: 0x264a0700
16:23:35:998 Announced to internet https://phoscon.de/discover
16:23:40:130 New websocket 172.30.32.1:44514 (state: 3) 
[16:23:41] INFO: Successfully send discovery information to Home Assistant.
16:23:51:115 start reconnect to network
16:23:56:365 try to reconnect to network try=1
16:23:56:381 Device protocol version: 0x010B
16:23:56:458 Current channel 15
16:23:56:475 CTRL ANT_CTRL 0x03
16:23:56:503 Device protocol version: 0x010B
16:23:56:552 Current channel 15
16:23:56:568 CTRL ANT_CTRL 0x03
16:23:56:593 NET ZDO start network status 0xC4
16:24:01:615 try to reconnect to network try=2
16:24:01:629 Device protocol version: 0x010B
16:24:01:701 Current channel 15
16:24:01:715 CTRL ANT_CTRL 0x03
16:24:01:743 Device protocol version: 0x010B
16:24:01:792 Current channel 15
16:24:01:808 CTRL ANT_CTRL 0x03
16:24:01:833 NET ZDO start network status 0xC4
16:24:06:614 try to reconnect to network try=3
16:24:06:629 Device protocol version: 0x010B
16:24:06:700 Current channel 15
16:24:06:720 CTRL ANT_CTRL 0x03
16:24:06:748 Device protocol version: 0x010B
16:24:06:808 Current channel 15
16:24:06:827 CTRL ANT_CTRL 0x03
16:24:06:850 NET ZDO start network status 0xC4`

[image]

[image]

[image]

Any idea what to do is greatly appreciated. At the moment, i'm sitting in the dark :(

Smanar, 2020-02-22:
You are "not in network" on your last screen.
You have same situation when you try on a windows machine (you will not have your devices, but perhaps you can connect the conbee) ?

veloc1ty, 2020-02-22:
Thank you, I tried that, seems like I am able to get into the mesh network, but I still cannot control anything. The devices show up as items, but when I try to turn them off or on / dim, it doesn't do a thing.

Tried a factory reset now to re-add everything, but that doesn't work as well. Maybe the stick is broken?
Really weird.

Smanar, 2020-02-22:
It's for that I m asking for windows, it s easier to make it working on it.
But if it works on windows, I don't think the device is broked.

On the last screenshoot, name are grayed, it mean they are not connected.

And if you still have "nichr verbuten" as firmware, deconz will not work too.

You can check if USB connexion is working with "GCFFlasher_internal -l"

veloc1ty, 2020-02-22:
The USB Connection is working, yes - I tried flashing the stick with the same firmware again which was successful.

Now I have the firmware showing up (for whatever reason), but I can only add one of my lights. The others are not showing up even though I resetted them properly with either the 5x 5sec on/off for my Lightify and the dimmer switch for Hue (they blinked). Still not showing up.

Resetted the gateway, resetted all lights, they show up in the mesh, the network says it is connected, but i am unable to talk to any of them. Tried USB extension / hub as well - and plugged it in to a VM on ubuntu and windows 10 with the same result..

Startup Log:

> starting version 237
> [19:08:42] INFO: Waiting for device...
> [19:08:43] INFO: Starting VNC server...
> [19:08:46] INFO: Starting the deCONZ gateway...
> libpng warning: iCCP: known incorrect sRGB profile
> 19:08:47:063 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
> 19:08:47:076 CTRL. 3.22.019:08:47:106 COM: /dev/ttyACM0 / serialno: DE2132713
> 19:08:47:106 COM: --dev: /dev/ttyACM0 (ConBee II)
> 19:08:47:106 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
> 19:08:47:138 parent process bash
> 19:08:47:138 gw run mode: docker/hassio
> 19:08:47:138 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
> 19:08:47:138 DB sqlite version 3.22.0
> 19:08:47:138 DB PRAGMA page_count: 30
> 19:08:47:138 DB PRAGMA page_size: 4096
> 19:08:47:138 DB PRAGMA freelist_count: 0
> 19:08:47:138 DB file size 122880 bytes, free pages 0
> 19:08:47:138 DB PRAGMA user_version: 6
> 19:08:47:138 DB cleanup
> 19:08:47:139 DB create temporary views
> 19:08:47:140 don't close database yet, keep open for 900 seconds
> 19:08:47:140 started websocket server at port 8081
> 19:08:47:141 found node plugin: libde_rest_plugin.so - REST API Plugin
> 19:08:47:142 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
> [19:08:47] INFO: Starting Nginx...
> [19:08:47] INFO: Running Hass.io discovery task...
> [19:08:47] INFO: Running the deCONZ OTA updater...
> 2020/02/22 19:08:47 [notice] 554#554: using the "epoll" event method
> 2020/02/22 19:08:47 [notice] 554#554: nginx/1.14.0 (Ubuntu)
> 2020/02/22 19:08:47 [notice] 554#554: OS: Linux 4.19.104
> 2020/02/22 19:08:47 [notice] 554#554: getrlimit(RLIMIT_NOFILE): 1048576:1048576
> 2020/02/22 19:08:47 [notice] 554#554: start worker processes
> 2020/02/22 19:08:47 [notice] 554#554: start worker process 566
> [19:08:47] INFO: Running the IKEA OTA updater...
> [19:08:47] INFO: Running the LEDVANCE/OSRAM OTA updater...
> [19:08:47] INFO: deCONZ is set up and running!
> 19:08:48:378 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
> 19:08:48:394 COM: /dev/ttyACM0 / serialno: DE2132713
> 19:08:48:394 COM: --dev: /dev/ttyACM0 (ConBee II)
> 19:08:48:429 created username: 56478788EA, devicetype: Home Assistant
> 19:08:48:448 Device firmware version 0x264A0700
> 19:08:48:452 unlocked max nodes: 200
> 19:08:48:550 Device protocol version: 0x010B
> 19:08:48:557 new node - ext: 0x00212effff050589, nwk: 0x0000
> 19:08:48:626 Current channel 11
> 19:08:48:646 CTRL ANT_CTRL 0x03
> 19:08:48:680 Device protocol version: 0x010B
> 19:08:48:743 Current channel 11
> 19:08:48:766 CTRL ANT_CTRL 0x03
> 19:08:49:090 New websocket 127.0.0.1:45118 (state: 3) 
> 19:08:52:222 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
> 19:08:52:222 GW firmware version: 0x264a0700
> 19:08:52:222 GW firmware version is up to date: 0x264a0700
> 19:08:52:994 Announced to internet http://dresden-light.appspot.com/discover
> 19:08:53:212 don't close database yet, keep open for 900 seconds
> 19:08:54:448 New websocket 127.0.0.1:45144 (state: 3) 
> 19:08:56:256 New websocket 172.30.32.1:48788 (state: 3) 
> 172.30.32.2 - - [22/Feb/2020:19:08:57 +0100] "GET /api/8B62569DED/sensors/1 HTTP/1.1" 403 96 "http://192.168.1.22:8123/api/hassio_ingress/pS58KaFt-VP3D7QFKbQSIC7UbYU5ck0jhBRE6WHYdbI/pwa/index.html" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/79.0.3945.130 Safari/537.36"
> [19:08:58] INFO: Successfully send discovery information to Home Assistant.

[image]

[image]

It does not discover any of the my other devices even though they are in pairing mode.
I have restored a deconz backup from 2 weeks ago with the same devices included, but it doesn't work either. I see them, but unable to control them or do anything with them.

veloc1ty, 2020-02-22:
It's just getting weirder. Got the Hue bridge from the basement and plugged it in, resetted some of the bulbs and was expecting to see them - didn't show up in Hue either.

I could add them using the serial number. Trying to get the conbee stick up and running again now (firmware not connected again..) and then trying to delete them from hue and add them back to deconz as for the initial installation which worked absolutely flawlessly. I really wonder what happend and what could cause ALL zigbee devices to stop working properly.

ebaauw, 2020-02-22:
> resetted some of the bulbs and was expecting to see them - didn't show up in Hue either.

Apparently the light didn't take the reset.  Note that for Hue lights, you need to continue to hold the _On_ and _Off_ buttons on the Hue dimmer, until the light burns steady.  If you release the buttons when the light starts blinking, the reset is canceled (or never takes place).

> I could add them using the serial number.

The Hue API takes a list of serial numbers in the body of the POST to `/lights`(to search for new devices).  I never looked at what ZigBee messages this translates to, but I think this "light stealing" is ZLL-specific magic.  I'm not sure if deCONZ could support this legally, as it might be based on the (once) secret ZLL keys.

> delete them from hue and add them back to deconz as for the initial installation which worked absolutely flawlessly

The Hue bridge factory-resets devices when removing them (see #2377), making them available for pairing.

> I really wonder what happend and what could cause ALL zigbee devices to stop working properly.

Over the past 4-5 years of my ZigBee experience, I never had to reset and repair my entire network.  I did have two or three scary moments, though, where the entire network became unresponsive (even when controlling lights directly from wireless switches).  I remedied these situations by powering down all routers, power cycling the RaspBee/ConBee (a reboot of the host is not enough), and slowly turning on the routers, one by one, waiting for each do be displayed in the GUI before powering on the next.  I haven't seen these issues since I replaced the problematic lights (with routing problems) by Hue lights.

veloc1ty, 2020-02-24:
Thank you, the hue dimmer reset trick was really helpful and helped me to rebuild everything.
However, I'm still wondering what might have happened to my former network. I digged through the logs but could not find any helpful information. There was no reboot or anything unusual, it just stopped working completely and I was not able to talk to any device using the settings which worked perfectly fine for 3 month or so.

closed 2020-02-24 by veloc1ty

Nebuchanezzer, 2020-05-24:
Same thing happened to me right now, I'm not able to join the mesh.

Just out of the blue...

Latest entries of the log i hassio:
13:41:37:209 CTRL ANT_CTRL 0x03
13:41:37:235 Device protocol version: 0x010B
13:41:37:284 Current channel 11
13:41:37:304 CTRL ANT_CTRL 0x03
13:41:37:332 NET ZDO start network status 0xC4
13:41:42:124 try to reconnect to network try=8
13:41:42:129 Device protocol version: 0x010B
13:41:42:184 Current channel 11
13:41:42:201 CTRL ANT_CTRL 0x03
13:41:42:229 Device protocol version: 0x010B
13:41:42:280 Current channel 11
13:41:42:298 CTRL ANT_CTRL 0x03
13:41:42:326 NET ZDO start network status 0xC4
13:41:47:123 try to reconnect to network try=9
13:41:47:138 Device protocol version: 0x010B
13:41:47:196 Current channel 11
13:41:47:214 CTRL ANT_CTRL 0x03
13:41:47:244 Device protocol version: 0x010B
13:41:47:301 Current channel 11
13:41:47:318 CTRL ANT_CTRL 0x03
13:41:47:346 NET ZDO start network status 0xC4
13:41:52:123 try to reconnect to network try=10
13:41:52:132 Device protocol version: 0x010B
13:41:52:185 Current channel 11
13:41:52:201 CTRL ANT_CTRL 0x03
13:41:52:229 Device protocol version: 0x010B
13:41:52:280 Current channel 11
13:41:52:299 CTRL ANT_CTRL 0x03
13:41:52:332 NET ZDO start network status 0xC4
13:41:57:124 reconnect network failed, try later
13:41:57:624 start reconnect to network
13:42:02:874 try to reconnect to network try=1
13:42:02:889 Device protocol version: 0x010B
13:42:02:945 Current channel 11

I've tried rebooted hassio, unplugged the stick.
Nothing helps...

Nebuchanezzer, 2020-05-24:
Here's the log right after restarting the plugin:

13:46:23:093 DB PRAGMA page_count: 40
13:46:23:093 DB PRAGMA page_size: 4096
13:46:23:093 DB PRAGMA freelist_count: 1
13:46:23:093 DB file size 163840 bytes, free pages 1
13:46:23:093 DB PRAGMA user_version: 6
13:46:23:093 DB cleanup
13:46:23:094 DB create temporary views
13:46:23:112 don't close database yet, keep open for 900 seconds
13:46:23:112 started websocket server at port 8081
13:46:23:114 found node plugin: libde_rest_plugin.so - REST API Plugin
13:46:23:115 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
[13:46:23] INFO: Starting Nginx...
[13:46:23] INFO: Running Home Assistant discovery task...
[13:46:23] INFO: Running the deCONZ OTA updater...
2020/05/24 13:46:23 [notice] 526#526: using the "epoll" event method
2020/05/24 13:46:23 [notice] 526#526: nginx/1.14.0 (Ubuntu)
2020/05/24 13:46:23 [notice] 526#526: OS: Linux 4.19.115
2020/05/24 13:46:23 [notice] 526#526: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/05/24 13:46:23 [notice] 526#526: start worker processes
2020/05/24 13:46:23 [notice] 526#526: start worker process 534
[13:46:23] INFO: Running the IKEA OTA updater...
[13:46:23] INFO: Running the LEDVANCE/OSRAM OTA updater...
[13:46:23] INFO: deCONZ is set up and running!
13:46:24:406 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
13:46:24:419 COM: /dev/ttyACM0 / serialno: DE1964520
13:46:24:419 COM: --dev: /dev/ttyACM0 (ConBee II)
13:46:27:898 New websocket 172.30.32.1:49970 (state: 3) 
13:46:28:889 Announced to internet http://dresden-light.appspot.com/discover
13:46:32:871 New websocket 172.30.32.1:58798 (state: 3) 
[13:46:33] INFO: Successfully send discovery information to Home Assistant.
13:46:34:134 COM: /dev/ttyACM0 / serialno: DE1964520
13:46:34:134 COM: --dev: /dev/ttyACM0 (ConBee II)
13:46:34:170 Device firmware version 0x26490700
13:46:34:174 unlocked max nodes: 200
13:46:34:250 Device protocol version: 0x010B
13:46:34:256 new node - ext: 0x00212effff042821, nwk: 0x0000
13:46:34:306 don't close database yet, keep open for 900 seconds
13:46:34:306 LightNode 2: Lekrum - Not - Upplampa added
13:46:34:330 don't close database yet, keep open for 900 seconds
13:46:34:331 LightNode 1: Lekrum - Not - Läslampa added
13:46:34:352 SensorNode 32 set node 0x00158d000252d7bf
13:46:34:353 SensorNode 33 set node 0x00158d000252d7bf
13:46:34:375 don't close database yet, keep open for 900 seconds
13:46:34:376 LightNode 3: Hall - Klabb added
13:46:34:394 don't close database yet, keep open for 900 seconds
13:46:34:394 LightNode 4: Vardagsrum - Hektar - Nere added
13:46:34:416 SensorNode 2 set node 0x00158d000238c9cc
13:46:34:416 SensorNode 3 set node 0x00158d000238c9cc
13:46:34:416 SensorNode 4 set node 0x00158d000238c9cc
13:46:34:434 SensorNode 5 set node 0x00158d000238c31a
13:46:34:434 SensorNode 6 set node 0x00158d000238c31a
13:46:34:434 SensorNode 7 set node 0x00158d000238c31a
13:46:34:456 SensorNode 8 set node 0x00158d000238c1ea
13:46:34:457 SensorNode 9 set node 0x00158d000238c1ea
13:46:34:457 SensorNode 10 set node 0x00158d000238c1ea
13:46:34:475 don't close database yet, keep open for 900 seconds
13:46:34:475 LightNode 5: Vardagsrum - Hektar - mitten added
13:46:34:498 don't close database yet, keep open for 900 seconds
13:46:34:498 LightNode 7: Vardagsrum - Aröd added
13:46:34:516 don't close database yet, keep open for 900 seconds
13:46:34:517 LightNode 8: Sovrum - Not - Uppljus added
13:46:34:540 don't close database yet, keep open for 900 seconds
13:46:34:540 LightNode 9: Sovrum - Not - Läslampa added
13:46:34:558 SensorNode 14 set node 0x00158d0002386d05
13:46:34:580 SensorNode 12 set node 0x00158d000238b388
13:46:34:580 SensorNode 13 set node 0x00158d000238b388
13:46:34:581 SensorNode 11 set node 0x00158d000238b388
13:46:34:598 SensorNode 20 set node 0x00158d0002300ca0
13:46:34:598 SensorNode 19 set node 0x00158d0002300ca0
13:46:34:598 SensorNode 21 set node 0x00158d0002300ca0
13:46:34:621 don't close database yet, keep open for 900 seconds
13:46:34:621 LightNode 10: Vardagsrum - Hektar - Uppe added
13:46:34:638 SensorNode 22 set node 0x00158d0002905856
13:46:34:661 SensorNode 23 set node 0x00158d00028f43eb
13:46:34:678 don't close database yet, keep open for 900 seconds
13:46:34:679 LightNode 11: Vardagsrum - On/Off - Mediabänk added
13:46:34:679 SensorNode 24 set node 0x00158d0003283774
13:46:34:679 SensorNode 25 set node 0x00158d0003283774
13:46:34:701 don't close database yet, keep open for 900 seconds
13:46:34:702 LightNode 12: Vardagsrum - Uttag Baslåda added
13:46:34:702 SensorNode 26 set node 0x00158d0003619441
13:46:34:702 SensorNode 27 set node 0x00158d0003619441
13:46:34:719 don't close database yet, keep open for 900 seconds
13:46:34:720 LightNode 14: Lekrum - Serveruttag - On/Off added
13:46:34:720 SensorNode 30 set node 0x00158d000361dd20
13:46:34:720 SensorNode 31 set node 0x00158d000361dd20
13:46:34:743 don't close database yet, keep open for 900 seconds
13:46:34:744 LightNode 13: Lekrum - Skrivbord - On/Off added
13:46:34:744 SensorNode 28 set node 0x00158d000327c920
13:46:34:744 SensorNode 29 set node 0x00158d000327c920
13:46:34:761 SensorNode 34 set node 0x00158d000253c6ec
13:46:34:761 SensorNode 35 set node 0x00158d000253c6ec
13:46:34:784 SensorNode 36 set node 0xec1bbdfffea307a8
13:46:34:801 don't close database yet, keep open for 900 seconds
13:46:34:801 LightNode 15: Klädkammare - Glob added
13:46:34:824 SensorNode 37 set node 0x680ae2fffe3e006f
13:46:34:841 SensorNode 38 set node 0x00158d00024e35bc
13:46:34:863 SensorNode 39 set node 0x00158d000283bc2b
13:46:34:948 Current channel 11
13:46:34:965 CTRL ANT_CTRL 0x03