dresden-elektronik/deconz-rest-plugin #2483: Device support - Leviton Dimmer Switch DL6HD

nacho319, 2020-02-23:
I would like to get support added for the Leviton DL6HD dimmer switch.  Here's all the capture from the Conbee II.
Here are the screen shots requested from the wiki; the cluster information is empty in my deCONZ display, so I didn't upload that.  I can if wanted/needed - but there is no cluster.

If there is anything else I can do, please let me know - happy to help.

[image]

[image]

[image]

Logfile from startup with device:

[deconz-22Feb2020-2-github.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4240580/deconz-22Feb2020-2-github.log)

nacho319, 2020-02-23:
[image]
[image]

nacho319, 2020-02-23:
[image]

ebaauw, 2020-02-23:
It’s already supported, by the look of things as `/lights/2`.  Did you install it as master (wired to a load) or as remote in a virtual three way setup?

I have no idea what the second endpoint does, you’d need technical documentation for that.  Or sniff the ZigBee traffic between the dimmer and the Leviton gateway.

nacho319, 2020-02-23:
The DL6HD comes through as a light, but its a switch.  

I can, using Apple Homekit (via Homebridge) / Phoscon adjust the dimming setting on the switch, but I don't have it installed as a master.  I would like to use it as a remote in a virtual switch configuration.

Am I missing something in how to do that setup wise?  I'm happy enough to just add another XML catalog and define the device as a switch.  (I haven't gone through the XML enough to learn all the structure, etc. - but I assume that's possible.)  If that might make the device appear as a switch and then allow Homekit/Phoscon to use it as a controller.

ebaauw, 2020-02-23:
> The DL6HD comes through as a light, but its a switch.

It's both, actually.  The master/wired part looks, feels, smells, and behaves like a light, hence it's exposed as a `/lights` resource.  The remote/wireless part behaves like a wireless switch, that should be exposed as a `/sensors` ZHASwitch resource.  The setup of `/lights` resources is pretty standard, and most devices are supported out-of-the-box.  The setup of `/sensors` resources is not, and devices need to be whitelisted for support.  Unfortunately in the C++ code, no XML configuration here.

Note that the server (blue) _On/Off_, _Level Control_, _Groups_, and _Scenes_ clusters make up the wired `/lights` stuff; the client (grey) _On/Off_ and _Level Control_ clusters make up the wireless `/sensors` stuff.  Not sure what the _Occupancy Sensing_ cluster is supposed to do.  Maybe to bind an occupancy sensor so that the dimmer turns on (off) automatically when occupancy is (no longer) detected.  Also no clue what the 0x8007 cluster on the seconds endpoint is for.  Probably some Leviton-specific stuff.  We'd need technical documentation from Leviton or to sniff the ZigBee traffic between the dimmer and the Leviton gateway.

Can you try and bind the **client** _On/Off_ and _Level Control_ clusters to a group, add a light to that group, and check that the dimmer controls the light?  Next step would be to add the dimmer to `supportedDevices[]`, turn on debug logging in deCONZ and see what commands it sends to the group.  Then, create a `ButtonMap[]` for the sensor to translate these commands to the appropriate `buttonevent` values.

nacho319, 2020-02-23:
Okay - this will be a nice trial-by-fire.  

Day 1: receive package and install pi & deconz
Day 2: work through understanding how device shows up
Day 3: tweak code to accept other devices

I found that there was some LeGrande dimmer switches that are roughly similar to this Leviton.  I'm going to see if I can follow that as a template for the code changes needed.

I understand why this shows up as both a light and a switch.  I didn't really think about it, originally, because I'm only trying to use this as a remote switch.  I don't have dumb light bulbs attached.  If I had the dumb bulbs attached, then this would make them into the smart bulbs, of course.  I have nothing wired to this switch as the load.

nacho319, 2020-02-29:
Erik, you said:
Can you try and bind the client On/Off and Level Control clusters to a group, add a light to that group, and check that the dimmer controls the light?

How do I do this binding?  I can't do it from Phoscon, and if there's a way to do it in deCONZ, someone would have to tell me how.

I tried to add it to supportedDevices[].  I basically went through the source code and read all the places where I saw the LeGrand dimmer switch and added a case for the Leviton switch.  I'm running that updated library locally now, but I'm not really seeing any behavior change.  

I presume that means I haven't done something useful with buttonMap[] yet?  Not sure how I know that yet.

I also need to learn how to debug deconz.  Are there any guides?

ebaauw, 2020-02-29:
> How do I do this binding? I can't do it from Phoscon, and if there's a way to do it in deCONZ, someone would have to tell me how.

In the GUI, see the user manual (under _Help_) and many examples in other issues.

> I tried to add it to supportedDevices[].

That in itself should be enough to create a ZHASwitch `/sensors` resources a for the client _On/Off_ and _Level Control_ clusters.  The rest is creating a group setting up bindings, configuring reporting for the battery etc.

> I presume that means I haven't done something useful with buttonMap[] yet?

That’s for mapping the commands that the (wireless function of the) dimmer sends to `buttonevent` values.

> Are there any guides?

Afraid not.  I did a fairly complete writeup for the Opple switches, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061.

nacho319, 2020-03-01:
> > How do I do this binding? I can't do it from Phoscon, and if there's a way to do it in deCONZ, someone would have to tell me how.
> 
> In the GUI, see the user manual (under _Help_) and many examples in other issues.
> 
Okay, I read the manual and did this, it didn't seem to work.  

side comment: the User Guide is a bit thin, but quite informative.

> That in itself should be enough to create a ZHASwitch `/sensors` resources a for the client _On/Off_ and _Level Control_ clusters. The rest is creating a group setting up bindings, configuring [...]

I'll be reading this issue now to see what I'm missing.  Thanks.

ebaauw, 2020-03-01:
> Okay, I read the manual and did this, it didn't seem to work.

That could very well be because the dimmer doesn’t control just any smart light, but only the associated master dimmer.  It might use a non-standard way to set this up, probably involving the 0x8007 cluster.  We need a sniffer log of how the Leviton gateway sets up the dimmer.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

colwellkr, 2021-08-01:
@nacho319 did you have any luck getting the dimmer to control other devices as a switch?

I just picked up the DG6HD (seems pretty much identical) hoping to do the same thing you were.
As a last resort, I can tell Home Assistant to sync the states of the lights, but doing it in Deconz/Phoscon would be much cleaner.