dresden-elektronik/deconz-rest-plugin #2485: IKEA Tradfri Remote (FW 2.3.014) automatic group is not selectable in phoscon

jan666, 2020-02-23:
I have 4 Tradfi Remotes. One is on firmware 1.2.223, the 3 others are on firmware version 2.3.014.

The 1.2.223 one works perfect, I can access the group on the left side in the phoscon app, assign it to lights an everything is perfect.

The three on 2.3.014 create groups on the left side in the phoscon app, but I cant enter them to assign it to lights.

Console logs (Chrome):

click on 1.2.223:

```
current site: group/lights
main.js?8603a4:3 goto: control/switch/<id>/group/37606
click on one of the 2.3.014:
```

```
current site: group/lights
goto: control/switch/<id>/group/undefined
```

```
pi@raspberrypi:~ $ dpkg -l | grep -i deconz
ii  deconz                             2.05.71-raspbian-buster-stable      armhf        A generic ZigBee monitoring and control tool
```

SwoopX, 2020-02-23:
Well, it looksed like I had a flank wide open related to groups, client clusters and their binding, pretty much due to my set up (barely use lights and switches, tldr...). While playing around in this regard the last couple of days, I pretty much had the same experience with other devices.

You can do what you're after manually in deconz GUI afaik. Also, I don't know the device, so bare with me. First step would be, under the assumption that the device you want to steer is not assigned to a group yet, to create a group for it. Select the groups cluster (blue) in deconz GUI and add a new group (ID is in hex). Now, the device to steer (e.g. a light) should have an on/off cluster, but eventually also a level control cluster. If so, keep that in mind. Next, get the binding dropbox from the "panel" menu.

Let's do the binding for on/of first. As source, drag the on/off client cluster (grey) from the remote. As destination, drag the endpoint from the device to steer containing the on/off server cluster (blue), but select "group" and enter the previously created group ID in hex. Repeat the steps, if both devices have a level control cluster.

@ebaauw happy if could add in or correct me where required.

ebaauw, 2020-02-23:
There's probably at least half a dozen issues discussing exactly this issue.  As I remarked earlier:
- ZLL switches (controlling devices) typically pick a random group on factory reset and send commands from their client clusters to that group out-of-the-box.  The only thing the REST API needs to do is pickup the group ID while eavesdropping  the commands, set `config.group`, and create the `/groups` resource;
- ZHA /ZIgBee 3 switches (controlling devices) don't pick a group.  Their client clusters need to be bound explicitly to a group.  If you're lucky, they don't send any commands out-of-the-boz, if you're unlucky the send broadcasts, controlling all lights.  For these switches, the REST API needs to pick a group, create the `/groups` resource, and setup the bindings from the client clusters to that group.  I think `config.group` is set on eavesdropping the commands.

You can tell what type your switch is from the endpoint in the GUI, or by looking at the value of `config.group`.  The groups picked by deCONZ for ZHA / ZB3 switches are numbered sequentially from 1; the groups picked by the ZLL switches have higher, random numbers.

The Trådfri remote was originally released as a ZLL switch.  With the latest firmware, it has changed to a ZHA/ZB3 switch.  The REST API plugin still treats it as a ZLL switch, and doesn't create the group.  Ideally, it should check the firmware version and adjust its behaviour accordingly.  Not sure how (where in the code) to do that, though, since the firmware version hasn't been read at the `/sensors` resource is created.

manup, 2020-02-24:
For the Phoscon App part in `main.js` @YKO-de made some changes which should improve it there. Will be available tomorrow-ish.

ebaauw, 2020-02-25:
The latest firmware for the Trådfri dimmer will turn it into a ZB3 device as well, see cross ref above.

ebaauw, 2020-02-25:
> Ideally, it should check the firmware version and adjust its behaviour accordingly. Not sure how (where in the code) to do that, though, since the firmware version hasn't been read at the time the /sensors resource is created.

Note to self: No, you ***, use the profile ID from the endpoint.

jan666, 2020-02-26:
Hi,

I just updated to 2.05.74 - now all 3 of my not-working groups are gone in the phoscon ui.

@ebaauw I can "confirm" what you said - the working remote is "ZLL", the non-workings are "HA"

[image]

[image]

@SwoopX I think I tried what you said but get the following error on "Bind":

[image]

ebaauw, 2020-02-26:
Yes, same issue.  If memory serves, the switch sends groupcast commands to group 0x0000, which the REST API handles like a broadcast, but the lights don’t.

jan666, 2020-02-27:
> If memory serves, the switch sends groupcast commands to group 0x0000

is it possible to remove that and assign the on/off cluster and level control cluster to a different group? I had no success so far

ebaauw, 2020-02-27:
Yes it is, simply bind these clusters to a group.  Make sure to press a button on the remote, so it’s awake, or it won’t receive the bind request.

I now have a Bitron remote which sends group broadcasts that are understood by a Hue light, but not by deCONZ.  I need to analyse the sniffer log to figure out wtf is happening here.  Looks like there’s more than one way to do group addressing in ZigBee?!

I’m trying to upgrade the firmware of my Trådfri dimmer, but it seems to stall.  This weekend, time permitting, I’ll have a go at changing the pairing logic for the remote and the dimmer, based on the profile (ZLL vs ZHA for ZB3).

jan666, 2020-02-27:
Is there a way to list all existing bindings? (deCONZ is confusing for newbies)

jan666, 2020-02-27:
> Yes it is, simply bind these clusters to a group. Make sure to press a button on the remote, so it’s awake, or it won’t receive the bind request.

I created a new group (it got id 15), bound the on/off cluster to the group 0x00f ("success"), added one light to that group but pressing the on-off-switch does not trigger the light

ebaauw, 2020-02-27:
If you control the group from Phoscon or thru the API, does the light react?

Does the sensor resource’s `config.group` attribute reflect the bound group?  You might need to bind the _Level Control_ and _Color Control_ clusters as well.

From memory, the remote sends _Toggle_ when pressing the _On/Off_ button.  Does your light support that command?  Check by issuing that command in the GUI on the server (blue) _On/Off_ cluster of the light.

jan666, 2020-02-27:
> If you control the group from Phoscon or thru the API, does the light react?

yes

> Does the sensor resource’s `config.group` attribute reflect the bound group? 

I dont really know what that means / how to check :(

> You might need to bind the _Level Control_ and _Color Control_ clusters as well.

Light an button only have on/off cluster and level control cluster. bound both - does not change anything

> From memory, the remote sends _Toggle_ when pressing the _On/Off_ button. Does your light support that command? Check by issuing that command in the GUI on the server (blue) _On/Off_ cluster of the light.

Yes, toggle works. (Its a IKEA FLOALT btw.)

ebaauw, 2020-02-27:
> Light an button only have on/off cluster and level control cluster.

My bad, it’s the _Scenes_ cluster.  Note that you need to bind the client (grey) clusters from the remote to the group.  The light only needs to be a member of the group.

> I dont really know what that means / how to check :(

Check the ZHASwitch `/sensors` resource for the remote thru the API.

jan666, 2020-02-28:
> My bad, it’s the _Scenes_ cluster. Note that you need to bind the client (grey) clusters from the remote to the group. The light only needs to be a member of the group.

There is no "scenes" cluster:

[image]

> Does the sensor resource’s config.group attribute reflect the bound group?

```json
{
  "config": {
    "alert": "none",
    "battery": 74,
    "group": "14",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "<etag>",
  "manufacturername": "IKEA of Sweden",
  "mode": 1,
  "modelid": "TRADFRI on/off switch",
  "name": "TRÅDFRI an/aus Taster",
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2020-02-28T06:28:23"
  },
  "swversion": "2.2.008",
  "type": "ZHASwitch",
  "uniqueid": "<uniqueid>"
}
```

group id 14 is the auto created group (invisible in phoscon)

```json
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "18"
  ],
  "etag": "757a79b56714ea88cf51f3c65a79d788",
  "id": "14",
  "lights": [],
  "name": "TRADFRI on/off switch ",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup",
  "uniqueid": "14:b4:57:ff:fe:c8:e1:b0"
}
```

my self created group is id 15:

```json
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [],
  "etag": "695fffed779d56b6c79204168d4b43bd",
  "id": "15",
  "lights": [
    "1"
  ],
  "name": "ON-OFF-Schalter",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```

stromnet, 2020-02-28:
Hey, just want to +1 on this and report that I have these weirdnesses as well.. As a new Conbee/ZigBee user I'm quite confused, I was under the impression that ConBee/ZB was mature enough that these kind of problems should not be present at all.. But great to see it's being worked on :)

Summary of my experience, partially from learning in this thread: The Remote was visible as this weird group that could not be clicked, until upgrade to 2.05.74 (dockerized). Then it disappeared. But still very confused with the groups in VNC vs in rest based UIs..

The On/Off switch with firmware 2.2.010 seems to allocate it's own group upon joining/reset, which is not visible in Phoscon. Trying to send "Bind" commands to the switch, to another group or destination, only results in a success but nothing actually working. The only success I've had is to use that self-assigned hidden group, and put the lights I want to control into those groups. Then it seemingly works. 
I've done all binding jobs in deCONZ UI (VNC), but the group changes there and the groups visibile on Phoscon/old web UI does not seem to match very well, adding to the confusion.

Unrelated to the on/off switch, I've managed to do the same (I think..? so many turns now..) on the round remote control with firmware 2.3.014. After adding that and a FLOALT panel to the conbee network, light level worked fine but not color. I tried binding the remote directly to the panel (holding reset until light started rapid flashes, i.e. normal pair) and now the center button works fine as well for that light. However, cannot control the colors via remote at all. Both light level & color temperature works fine from Phoscon though. Tried binding the "Level control" cluster on the remote to both the group, and directly to the FLOALT, but no joy.

Going through Conbee rules/mappings/whateverits'called is surely possible but a non-acceptable workaround, as these basics should "just work" imo.

Seems like I picked a bad time to jump on the TRÅDFRI train, ZB3 upgrades on devices which causes SW incompatabilities.. Wellwell, looking forward to further updates and hassle free experience :)

ebaauw, 2020-02-28:
@jan666 you’re showing an on/off switch, not the 5-button remote that this issue is about.

jan666, 2020-02-28:
@ebaauw yea, the 5-button-remote works fine after the bindings (remainng issues: its not possible to do in phoscon, some groups are now invisible in phoscon after the bindings. AFAIK the "color control" of the remoted is non-standard, so it does not work. as the group is not invisible in phoscon I cant add custom actions for that), I never had the "broadcast" issue there. Only on the on/off switch. My last questions were all releated to the on-off-switch (ZB3 firmware), sorry for not making that clear.

I did what @stromnet did and assigned my FLOALT to the auto-created on-off-switch group (id: 14) and changed on/off-cluster and level control-cluster to 0x00e. Now the on-off-switch works with my FLOALT. I removed all bindings from the on-off-switch (both clusters) to 0x000e and 0x000f but it still works. now the `config.group` of the on-off-switch changed to 13 (a group that I created myself in phoscon):

```json
{
  "config": {
    "alert": "none",
    "battery": 87,
    "group": "13",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "<etag>",
  "manufacturername": "IKEA of Sweden",
  "mode": 1,
  "modelid": "TRADFRI on/off switch",
  "name": "TRÅDFRI an/aus Taster",
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2020-02-28T07:22:19"
  },
  "swversion": "2.2.008",
  "type": "ZHASwitch",
  "uniqueid": "<uniqueid>"
}
```

```json
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "17",
    "18"
  ],
  "etag": "<etag>",
  "id": "13",
  "lights": [
    "1"
  ],
  "name": "Büro",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```

Everythning is a little bit confusing

ebaauw, 2020-02-28:
> the 5-button-remote works fine after the bindings (remainng issues: its not possible to do in phoscon, some groups are now invisible in phoscon after the bindings

I don't know (nor use) Phoscon, but the old web app uses a special header in the REST API request to unlock some additional functionalities.  One of these is the `hidden` attribute to a group.  Of course, I added support for this to `ph`: e.g. use `ph -p get /groups` to see it.  
I did some investigations on my test system, but `hidden` is not set for any of the groups, and Phoscon doesn't show them anyways.  I suspect Phoscon hides the group when `devicemembership` is non-empty, under the (possibly correct) assumption that this is too confusing for end-users.  It creates rules when assigning actions to the corresponding device, which could couteract the group commands.  
So to assign lights to the group:
- Use the old web app; or
- Use the API (`ph put /groups/1 '{"lights": ["1", "2", "3"]}`; or
- Use the GUI to add the group to the light (!), in the _Cluster Info_ panel for the server (blue) _Groups_ cluster of the light.

> AFAIK the "color control" of the remoted is non-standard, so it does not work.

The remote sends manufacturer-specific _Scenes_ commands on the _Next_ and _Previous_ buttons.  These commands are only understood by Trådfri lights - in their factory new state.  On pairing, the REST API plugin adds the light to `/groups/0`, which causes the lights no longer to react to the manufacturer-specific commands.  
Of course, you can still create rules on the gateway for the corresponding button events, but they won't work when the gateway is down.

> I never had the "broadcast" issue there.

Probably because your remote was paired while on old firmware.  I just upgraded my Trådfri wireless dimmer to ZB3 using the Trådfri hub (in deCONZ, the upgrade stalled at 98%).  After resetting the dimmer and re-pairing it to deCONZ, it restored the old group bindings.  If you reset the remote, delete the `/sensors` resource, and re-pair the remote, you'll probably see the broadcast issue.  Not that I'd recommend this - it's not particularly pretty ;-)

jan666, 2020-02-28:
Its really a little bit strange. I somehow manged to have a group:

```json
{
  "action": {
    "bri": 254,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "17",
    "18"
  ],
  "etag": "450e99db612fde988eaeffe797f41fb4",
  "id": "13",
  "lights": [
    "1"
  ],
  "name": "Büro",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```

`17` is my TRADFRI remote control (Name: "FB Büro", `18` my TRADFRI on/off switch. In this state its listed in phoscon but called "FB Büro" (like the remote control). The light reacts to both "sensors".

When I remove light 1 from the group via

`curl -vv -XPUT http://192.168.42.22/api/<key>/groups/13 -H "Content-Type: application/json" --data '{"lights": []}'`

It becomes (as expected):

```json
{
  "action": {
    "bri": 254,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "17",
    "18"
  ],
  "etag": "f4a68ba1e1e4c6930228e3242a0ee15b",
  "id": "13",
  "lights": [],
  "name": "Büro",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```

...but is now hidden in Phoscon.

I have another group called "Büro" (ID 1):

```json
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "21"
  ],
  "etag": "225874a7f49d8a6fbbbca5a8b71b66ba",
  "id": "1",
  "lights": [],
  "name": "Büro",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```

"Sensor" is another 5-button remote (called "TRÅDFRI Fernbedienung"). This is hidden in Phoscon. When I add light 1 to this group via

`curl -vv -XPUT http://192.168.42.22/api/<key>/groups/1 -H "Content-Type: application/json" --data '{"lights": [ "1" ]}'`

it becomes (as expcted):

```
{
  "action": {
    "bri": 254,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": true,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "21"
  ],
  "etag": "0b056ec8cb4fa1170ba82a816dfc4a6b",
  "id": "1",
  "lights": [
    "1"
  ],
  "name": "Büro",
  "scenes": [],
  "state": {
    "all_on": true,
    "any_on": true
  },
  "type": "LightGroup"
}
```

... is listed in phoscon as group(!) "TRÅDFRI Fernbedienung" (like the remote is called) - but the remote does not trigger the light.

ebaauw, 2020-02-28:
That would seem like a bug in Phoscon that it doesn't show empty groups ("empty" as in: not containing any lights).

> but the remote does not trigger the light.

Please understand that there's no centrally managed group object in ZigBee.  A group in ZigBee is just a multicast address.  To receive messages sent to this address, lights (and plugs, wired swiches, etc) need to "subscribe" to this address, by adding the group to their configuration (thru the _Groups_ server (blue) cluster).  To send messages to this address, wireless switches (and sensors acting like switches) need to bind the client (grey) cluster to the group.

The `/groups` resource is just cached information by the REST API plugin.  I suppose under rare circumstances it could become out of sync with the actual configuration (in non-volatile memory) on each light and wireless switch.  Double-check `config.group` on the `/sensors` resource to know what group the last command was sent to.  If you bind different (client) clusters to different groups, `config.group` will change with each command.  If you bind a client cluster to multiple groups, I suppose the (random) order in which deCONZ sees the commands will drive `config.group`.

jan666, 2020-02-28:
@explanations big thanks for you explanations. I might understand 5-10 % now, but I can somehow do want I wanted. Is there a good resource for learning more?

(and btw: is there a way to give a coffee to you for your help?)

ebaauw, 2020-02-28:
> Is there a good resource for learning more?

Afraid not.  I learned most from bitter experience over the last three years, by (in order of increasing difficulty):
- Reading the deCONZ user manual (under _Help_ in the GUI);
- Reading the Hue developer site, https://developers.meethue.com/develop/get-started-2;
- Sniffing the ZigBee traffic through ZShark (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/405);
- Trying to read the official ZigBee specs (obtainable at www.zigbee.org, after registration);
- Trying to understand the REST API code;
- Reading the documentation of the deCONZ C++ API for plugins (link is broken, see #2466, but the `deconz-dev` package installs the header files in `/usr/include/deconz`).

ebaauw, 2020-02-28:
Above commits create a new group and corresponding bindings for the Trådfri remote control and the Trådfri wireless dimmer running the new ZB3 firmware.  No more "broadcast" issues.

xxLeoxx93, 2020-04-26:
Is the fix already in the latest deCONZ release? Or do I need to install this (somehow) manualy?

xxLeoxx93, 2020-05-24:
Just wanted to report @ebaauw that it seems that the fix isn't working. My tradfri remote is still not able to change the color temperature of my lightbulb, however the left/right button press is recognized and changing the color temperature in the webinterface is also working. (2.05.76 & 26580700)

ebaauw, 2020-05-24:
> it seems that the fix isn't working. My tradfri remote is still not able to change the color temperature of my lightbulb

It might not do what you had hoped it would do, but that has nothing to do with the fix.  I never claimed to have fixed that.  In fact, it cannot be fixed: The remote sends manufacturer-specific _Scenes_ cluster commands on the left/right buttons.  These commands are only honoured by IKEA lights in factory-new state.  When pairing the lights to deCONZ, it adds them to the default group (for `/groups/0`), thereby invalidating the factory-new state.

If you want to use the left/right buttons, you need to create rules acting on the corresponding button events, or an API client app to create those rules for you.  That's beyond the scope of the REST API plugin.

xxLeoxx93, 2020-05-24:
Sorry, then I missunderstood your fix!
When you talk about rules: Are you talking about the swich editor in the phoscon app? Because there I assigned the Left/Right button to change the color temperature, but it doesn't do anything. What got me confused is that in the switch editor it detects when I press left/right, but it's not excecuting it.

ebaauw, 2020-05-24:
I read German alright, but I don't use Phoscon.  I think it's supposed to create these rules for you, but I'm not sure.

xxLeoxx93, 2020-05-24:
Alright, then'll create a seperate issue for that, since it has nothing to tho with this issue from the opening post. Thanks anyway!

Mimiix, 2020-06-12:
Moving this to Phoscon git as it was originally a phoscon question. Not sure if it is solved or not.