dresden-elektronik/deconz-rest-plugin #2486: Not able to add Deconz to Harmony Hub on Home-assistant

D3nnisd, 2020-02-24:
Today i learned you are able to add Deconz to Harmony as a Hue Bridge. However, my Harmony doesn't see it.

I configured Deconz on Home-Assistant in such a way (port 80), that i can use the Hue app and other apps.  

Is this new since the last versions? or is this Home-Assistant related?

alex0411, 2020-02-24:
I also noticed this: A few versions back I was able to control deconz from my Harmony. Deconz is still added as Hue bridge in harmony, when I try to readd it, the Harmony Hub cannot find it.

alex0411, 2020-02-27:
Are you using deconz in a docker Container?

D3nnisd, 2020-02-28:
Yes i am. Using it with Home Assistant.

alex0411, 2020-02-28:
Me too and I noticed this issue since I migrated to the docker environment. Could this be related? I will try to verify it

ldplusse, 2020-04-14:
I have the same problem. I am just using DeCONZ with ConBee on Windows 10.
My Harmony Companion cannot find any Hue bridge.

Mimiix, 2020-06-02:
Any news on this?

w00druff, 2020-06-09:
I was only able to add Deconz to my harmony hub after switching Deconz to port 80.
Works with Deconz v2.05.75, 76 and 77 here.

alex0411, 2020-06-09:
> I was only able to find Deconz with my harmony hub after switching Deconz to port 80.
> Works with Deconz v2.05.75, 76 and 77 here.

Hi,
What do you mean exactly with port 80? I am also running my deconz on port 80 (deconz is running in a docket container) and I am not able to add it to my harmony hub.

w00druff, 2020-06-09:
Exactly what i mean.
Im not really concern with docker but is your Deconz container configured as bridge, host or macvlan network ?

Mimiix, 2020-06-09:
I did some testing myself. 

I am running a VM with HA Generic linux install. the VM is within my DS218+ Synology nas. I run deCONZ with the addon.

In that case, the UPNP device sniffer didn't find it. 

I installed a second VM, installed clean ubuntu and installed deconz there. 

Now, my Harmony can find it. The issue is within either the Marthoc container or within the HA addon as i can use it within my HA.

alex0411, 2020-06-09:
> Exactly what i mean.
> Im not really concern with docker but is your Deconz container configured as bridge, host or macvlan network ?

I have tested it with bridge and host network. With bridge I cannot find it at all, with host I am able to find it, but I cannot add it.

@Mimiix : Could you also add it? Because I can find deconz with my harmony hub, but I cannot add it. When I try to connect, nothing happens.

Mimiix, 2020-06-09:
I can. I can provide a screen on discord.

w00druff, 2020-06-09:
But when you find it you also start 60sec pairing mode at the Deconz Webinterface right ?
You should try macvlan, only with this network mode your container gets handled like a normal device in lan environment.

Mimiix, 2020-06-10:
@w00druff I don't use a container. I installed the deconz package from apt.

After that, i told Harmony to search for a hue bridge. Then it asked me to press the Authenticate button in Phoscon.

Then it imported :)

w00druff, 2020-06-10:
My message was related to alex0411 posts :-)

alex0411, 2020-06-11:
> But when you find it you also start 60sec pairing mode at the Deconz Webinterface right ?
> You should try macvlan, only with this network mode your container gets handled like a normal device in lan environment.

Yes, I start the pairing mode and nothing happens.
I will try macvlan...

alex0411, 2020-06-12:
Today I tested my setup with docker and macvlan: Now I am able to add deconz as bridge to the harmony :-)

w00druff, 2020-06-12:
Great !

Mimiix, 2020-06-26:
As this is my own issue and seems to be resolved for others. I'll close this on my own behalf. For me it is solved!

closed 2020-06-26 by Mimiix