dresden-elektronik/deconz-rest-plugin #2487: Xiaomi Aqara door/window sensors not being added

criskolkman, 2020-02-24:
Hello,

I've seen the same issue a few times but I haven't found the solution.
I bought a ConBee II USB stick which I attached to a Windows 10 pc using an extension cable (about 2 meters).

The Xiaomi Aqara door/windows sensors (new sensors, never been used, bought on Ali Express) are NOT being discovered in the Phoscon web app or the deconz app.
I've tried to reset the sensors many times, holding them close to the USB stick, further away, using different channels in the Phoscon web app, nothing seems to help to connect them to the app.

What could be wrong here?

criskolkman, 2020-02-24:
Looks like that after turning off Windows firewall the sensors can be added (tested 1 now).
Will test the others and close this issue.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix