dresden-elektronik/deconz-rest-plugin #2489: Device support - Blitzwolf temp and humidity sensor BW-IS4

Valcob, 2020-02-25:
Hey and thanks in advance for a great job!
I would like to get support for this small temp and humidity sensor from Blitzwolf the [BW-IS4](https://www.banggood.com/custlink/vKmvUY5pLF)
[image]
It's totally fine discover-able by the  Deconz and I would like it to be passed to UI and my instance of home assistance

Basic Cluster:
[image]
Blitzwolf by the way what the hell is this naming though??

Power Cluster:
[image]

Temperature Cluster:
[image]

Humidity Cluster:
[image]

Please let me know if there is some thing else I need to post or add may be

SwoopX, 2020-02-25:
Already supported with release .74 from today.

Valcob, 2020-03-01:
Tested, It's working thanks !!!

closed 2020-03-01 by Valcob

srtxema, 2021-01-01:
i can´t sync with phoscon. is there any special sync mode?