dresden-elektronik/deconz-rest-plugin #2490: Repository corrupt?  Lines disappeared from light_node.cpp

ebaauw, 2020-02-25:
@manup, I'm missing the following lines in `light_node.cpp`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/a3e415d5fc6f73f602b6b6fcd04840bc9ef4f421/light_node.cpp#L542-L544

I have no clue what could have happened here: they were present in https://github.com/dresden-elektronik/deconz-rest-plugin/commit/a3e415d5fc6f73f602b6b6fcd04840bc9ef4f421 and they're not present in the current version of the file.  Looking at _Blame_, the lines weren't deleted in a later commit; it would seem they were simply never there?!  Is this a corruption in the repository?

SwoopX, 2020-02-25:
Crap, could have been my fault. Had to resolve merge conflicts and probably missed that one. Sorry for that!!

ebaauw, 2020-02-25:
Ah, OK.  Still weird that the lines don't appear as being deleted.

closed 2020-02-25 by ebaauw