dresden-elektronik/deconz-rest-plugin #2493: Bricked ConBee II after firmware update timeout

ReeChip, 2020-02-26:
after I tried to update with `wget http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x264a0700.bin.GCF` following happend: 
```$ sudo GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x264a0700.bin.GCF  

GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 264A0700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 158985 bytes: |==============error: timeout flashing firmware after 3007 ms
```
now I insert the ConBee II and it disconnects/connects after some seconds on Debian and Windows

manup, 2020-02-26:
The device can't be bricked, since the bootloader is static.

Can you please try with the following call:

    sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60  -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x264a0700.bin.GCF

ReeChip, 2020-02-26:
Thanks for that quick reply. I will try to flash it during the "conneted" state. 

Maybe I misunderstood something, the problem is on both my linux (debian), that I can see the device /dev/ttyACM0 and then it disappears:
I connect the device and spammed:
```
$ ls /dev/ttyA*
/dev/ttyACM0
$ ls /dev/ttyA*
ls: cannot access '/dev/ttyA*': No such file or directory
$ ls /dev/ttyA*
ls: cannot access '/dev/ttyA*': No such file or directory
$ ls /dev/ttyA*
ls: cannot access '/dev/ttyA*': No such file or directory
$ ls /dev/ttyA*
/dev/ttyACM0
```
etc... In windows I have the same issue. Device Manager keeps refreshing every couple seconds, additionally to connect/disconnect notification sounds.

manup, 2020-02-26:
hmm on Linux, can you please try to:

* Unplug the ConBee II
* Start the above command (with -t parameter)
* Plugin the ConBee II again

manup, 2020-02-26:
Are there any other USB devices connected (on Linux)?

ReeChip, 2020-02-26:
> Can you please try with the following call:
> 
> ```
> sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60  -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x264a0700.bin.GCF
> ```
with -t parameter fixed it! 

I got following Output:
```
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
unlock!  (148 ms)
flashing 158985 bytes: |=======error: timeout flashing firmware after 3003 ms
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 158985 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```

Thanks!

closed 2020-02-26 by ReeChip

dan003400, 2021-09-28:
Same issue, this device has yet to work properly, and now I can't even manually update this damn thing. What a piece of garbage.

Mimiix, 2021-09-29:
@dan003400 Not sure what you did but feel free to open a issue on the forum and i'm happy to ask people at DE to help you out :)

Johan-de-Bruin, 2022-07-04:
I recently had the same kind of problem: a failed firmware upgrade, and aftewards a connecting and instantly disconnecting Conbee II. 

I tried updating it with the flasher on several Windows 10 machines GCFFlasherl -d comportnumber -t 60  -f deCONZ_ConBeeII_FIRMWAREVERSION.GCF   But to no avail.

I even tried it with -t 3600  to have the flasher retry it for an hour (!) but nope.  Nothing.

If I gave a GCFflasher -l  ( list ) command, just after a connected sound, I could see the output with a serialnumber. 

Than I added the -sn deSERIALNUMBER to the command, and it flashed the conbee 2 at the first try!!

So first: GCFFlasher -l  to see the serialnumber than

Flash it with the new firmware using the following command.
(Replace X with COM number, Y with firmware filename, Z with the serialnumber) 

GCFFlasher -d COMX -f Y -sn Z

Smanar, 2022-07-04:
Ha ?
I didn't know it was possible ...
I will add it on the Wiki, can be usefull, thx a lot.

tiimonn, 2024-03-25:
I've had similar issues but as it seems the latest version does not have the -sn command line option.

manup, 2024-03-25:
Can you please try https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#what-if-conbee-all-versions-still-refuses-to-flash

Technically the ConBee can't be bricked since the bootloader is fixed, but if the application firmware is not running or corrupted flashing needs a bit of manual help by doing the reboot into bootloader by hand (the -sn parameter isn't needed here).

tiimonn, 2024-03-25:
> Can you please try https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#what-if-conbee-all-versions-still-refuses-to-flash
>
> Technically the ConBee can't be bricked [...]

I managed to flash the firmware but wanted to mention that the docs suggest a cli argument that is not supported (anymore).
The docs should be updated accordingly