dresden-elektronik/deconz-rest-plugin #2494: OTAU issues

realwax, 2020-02-26:
Maybe reopen this, please https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1217

Currently trying to update my Tradfri E14 to 2.3.007. It is a struggle. Sometimes it works, sometimes not. And if, just for two bulbs. It seems a simple deconz restart does not help. A reboot does sometimes. (Firmware?)  A query action done in the GUI on a bulb not being updated yet, does not cause any reaction or even debug with "deCONZ --dbg-otau=2"

[image]

Thank you for helping furhter/fixing!

realwax, 2020-02-26:
After a reboot, it works againg for 2-3 bulbs. Then a deconz restart might work for another run but then only a reboot works. After a reboot:

[image]

SwoopX, 2020-02-26:
Noticed that behaviour as well while updating my ubisys devices recently. First one went ok, second didn't felt to start. Couple of days and reboots later, it worked out. Kinda disturbing in my "prod" environment.

Mimiix, 2020-06-05:
I can confirm this still excists in my own environment. Keeping it open if others join.

realwax, 2020-06-26:
Issue not stale - simply not addressed yet ;)

MattWestb, 2020-07-19:
Fix for the 2.05.78 ?

manup, 2020-07-20:
How do you update the bulbs? The easiest path is to have the OTA files in `~/otau` directory and let the plugin do its work automatically.

So for example put the files in `~/otau` restart deCONZ, and power-cycle a bulb (or wait for a couple of hours). After power-up the bulb asks the OTA server for new version and the update should begin.

realwax, 2020-07-20:
Via GUI query and update as showed above in the screenshots. As sometimes the update freeze as well maybe due ikea bulbs maybe deconz your described path isn't really reliable as only some bulbs will be updated over time. In case of a rollback the manual method is required as well...
Thank you 🙏