dresden-elektronik/deconz-rest-plugin #2496: Device Support  Request - Sercomm / Telstra (SZ-ESW01-AU) Smart Plug

user2202, 2020-02-27:
Hello, I would like to get this smart plug fully supported. I can currently turn it on and off. I would also like to also get the electrical measurements.  Thank you
[image]

Basic Cluster
[image]

Node Info
[image]

Electrical Measurment - Power off
[image]

Electrical Measurment - Power on with device
[image]

SwoopX, 2020-02-27:
Can you also please provide a screenshot from the simple metering cluster? Thanks.

ebaauw, 2020-02-27:
According to the multipliers and divisors, the power is 58 * 128 / 1000 = 7.4W; the voltage is 30941 * 8 / 1000 = 247.5V; and the current is 54 * 1 / 1000 = 0.0054A.  That doesn't quite compute.  Also note the difference is reported frequency 50Hz without load and 47Hz with.  I have no clue what the _Power Factor_ means.

The lifetime (since last reset) consumption should be reported through the _Simple Metering_ cluster, usually also including the current power.

user2202, 2020-02-27:
Sorry I missed that one :)

The load I used to get the reading was 240V 0.5A 50Hz, I tested the same load on another SZ-ESW01-AU connected to smartthings and it gave a reading 6.3W.

I can re try with another powered device if needed.

[image]
[image]

SwoopX, 2020-02-27:
@user2202 Thanks, that's fine. Just wanted to be sure.

@ebaauw Also thanks. Seen the divisors and multipliers, but just wanted to double check. First time I've seen it so detailed. Now for the current consumption, the value is already in Wh. For the instantaneous demand, division by 1000 would fit the picture, right?

ebaauw, 2020-02-27:
You might need to decipher the _Summation Format_ (0x0303) and _Demand Format_ (0x0304) to know how to scale the values for _Current Summation Delivered_ and _Instantaneous Demand_.  The format fields consist of subfields, which the GUI cannot handle.  See the ZCL spec for the details (I don't know them by heart).

SwoopX, 2020-02-28:
Ah yes, forgot about that. So 0xCE is 11001110 and 0x96 is 10010110. For both values, the spec says: 

Bits 0 to 2: Number of Digits to the right of the Decimal Point.

In both cases, that would be 110 (starting the count from LSB), which is 6, resulting in 0,000661kWh for current summation. That smells a bit fishy and kinda imply the device has just been put into use.
For instantaneous demand, it'd be ‭0,008106‬kW -> 8,1W, which would come at least near the load applied.

@ebaauw Any mistake I made here?

ebaauw, 2020-02-28:
> That smells a bit fishy and kinda imply the device has just been put into use.

Just been reset.  Maybe for pairing with deCONZ?

meremortals70, 2020-04-28:
Is it possible these powerplugs could be moved to a "switch" instead of a light?

meremortals70, 2020-05-11:
What information do you need to enable these plugs to be calibrated correctly please?

They are currently registering negative values.

I have a power meter.  How many data points do you need and do you need resistive only or happy with reactive loads?

[image]

SwoopX, 2020-05-11:
Can you let me know on which deconz version you're running? Is your feedback based on checking the REST API? I also need to understand if that values you're referring to come from the ZHApower or ZHAconsumption sensor... I'd assume the ZHApower (how much power is consumed right now, not historically.)

meremortals70, 2020-05-11:
I’m not using ZHA, I am using the Deconz 2.05.75 integration to HA.

SwoopX, 2020-05-11:
I'm not referring to ZHA you mean. ZHApower or ZHAconsumption, that is the type of sensor created by deconz. I need to understand which one of them is producing the funny values.

meremortals70, 2020-05-11:
Ah sorry, ZHApower

SwoopX, 2020-05-11:
ok, thanks. That's weired. Can you provide just another screenshot from the electrical measurement cluster with the attributes read?

The point is, the rules of calculation are given there by the power multiplier and divisor. That's why I don't understand it.

meremortals70, 2020-05-11:
Sure, here you go.

[image]

SwoopX, 2020-05-11:
Thanks. So the calculation based on the screenshot should be 0x050B * 0x0604 / 0x06045, so 8 * 128 / 1000 = 1W

Would that make sense in that case?

meremortals70, 2020-05-11:
So we can be sure I have plugged an electric kettle into the power meter and then into the plug.

1st image is with the kettle on and not boiling. It is one of those fancy ones which provides current temperature etc of the water in the kettle.

228.9V
50Hz
Power Factor 0.2
0.01A
9 watts

[image]

The second image is with the kettle boiling
230V
50Hz
Power Factor 1.0
9.121A
2138W.

[image]

SwoopX, 2020-05-11:
For the 1st screenshot, the values result in 1,563W
For the 2nd, this is a better fit with 2033,152W

Also, the result cannot be negative, unless the plug reports some negative values here.

You are aware that power is also exposed via the consumption sensor here?

meremortals70, 2020-05-12:
In that case could it be Phoscon causing the negative values?

SwoopX, 2020-05-12:
So the values at the bottom indeed are correct. Voltage is provided in V and current is provided in mA, so it sould show around 91W power usage at that point in time. However, the datatype is signed integer, so negative values are allowed and valid in principle...

There's also a consumption sensor created which should provide power. Does it look more reasonable? Eventually, you must apply the calculation as above.

meremortals70, 2020-05-12:
Whilst it looks more reasonable it is cumulative

SwoopX, 2020-05-12:
It should contain 2 values, power (again) and consumption.

meremortals70, 2020-05-12:
Not exposed in the Deconz integration in Home Assistant as a sensor however I can see it on the Consumption Graph in the location where current and voltage is shown on the Power Graph. However, it is showing less than the Power Sensor. At 90w multiplying current and voltage, the power sensor says 4w and the power reading under the consumption sensor is 2w

SwoopX, 2020-05-12:
> At 90w multiplying current and voltage, the power sensor says 4w and the power reading under the consumption sensor is 2w.

So that's pretty useless either.

meremortals70, 2020-05-27:
Just keeping this issue open until I get time on the weekend for some quality diagnostics

SwoopX, 2020-05-27:
Sure. However, I tried to peek at z2m and SmartThing and they both don't have a solution :/

meremortals70, 2020-07-05:
Yep Deconz is not longer in my network due to the lack of support

closed 2020-08-02 by stale[bot]