dresden-elektronik/deconz-rest-plugin #2497: Device support request - Elko 316 GLED RF dimmer

uphillbattle, 2020-02-27:
https://proff.elko.no/getfile.php/13124115/produkter/doc/EKO05821/Datasheet4523410.pdf

This dimmer can be paired and controlled using the ConBee stick. 

Pairing is a pain - takes many tries - but eventually works.

Controlling the dimmer works, but setting the brightness at the same time as switching on or off does not work. For example, if the dimmer is off, it does not turn on if you set the brightness (to, for example, 50 %). So, if it is off, you first have to switch it on, then set the brightness.

Node info:
[image]

Clusters:
[image]

Basic cluster:
[image]

Level Control cluster:
[image]

ebaauw, 2020-02-27:
Most likely, the dimmer’s firmware doesn’t support _Move to Level with On/Off_.  See #2475.

uphillbattle, 2020-02-27:
Yes, I belive that's the case. I read the thread you referenced [#1111](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1111#issuecomment-456464872). Were you able to make the ConBee send on and brightness commands sequentially? If so, it could be done like this for the Elko dimmer (if it was supported). It's what I'm currently doing with an automation in Home Assistant, so for controlling the dimmer alone from HA, I've got it working, but I can't use "scenes" in HA since there I can't make it send command sequentially.

Mimiix, 2020-06-05:
@uphillbattle Is this  implemented?

uphillbattle, 2020-06-05:
I'm not sure. Maybe @ebaauw knows.

ebaauw, 2020-06-05:
> Were you able to make the ConBee send on and brightness commands sequentially?

Yes, but it's causing some other issues, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2824#issuecomment-635883214.

closed 2020-07-11 by stale[bot]