dresden-elektronik/deconz-rest-plugin #2498: Hourly ghost events Samjin door/window contact causing false tamper state

jcallaghan, 2020-02-27:
Similar to ["Ghost events with Aqara vibration Sensor #2128"](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2128) I am seeing regular "vibration" events from my Smartthings sensors. I have reset sensor and changed the battery to troubleshoot however this is occurring on 20+ sensors.

I'm seeing an active state triggered usually in 60-minute intervals or 60 minutes from the last state change. These sensors form triggers in Home Assistant so I've had the automation disabled since I moved over to my Conbee II. 

The Conbee II is connected to deCONZ running in docker. I have 109 nodes made up of battery-powered sensors and 66 powered lights. 

## deCONZ 
[image]

[image]

[image]

## Home Assistant
[image]

[image]

[image]

It feels like the device is waking up and causing a false tamper alert. Can I share a log or anything from deCONZ to help troubleshoot this?

Off-topic. Any other pointers or tips I should be considering with this network?

marcelodavanzo, 2020-02-28:
This is discussinf the same issue, unfortunately it doesn't look like there's any interest in fixing this.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1848

jcallaghan, 2020-03-01:
@ebaauw has kindly reviewed at the code and with the help of the device will look at this further. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1848#issuecomment-593000538

Mimiix, 2020-06-05:
As it's a duplicate, this issue will be closed. However, i've assigned a label to #1848 to prioritize it.

closed 2020-06-05 by Mimiix