dresden-elektronik/deconz-rest-plugin #2500: Phoscon shows Conbee Coordinator as Light

realwax, 2020-02-27:
After naming the Coordinator in the GUI, Phoscon displays the coordinator as light with delete option...

[image]

jan666, 2020-02-28:
Same here. I think this was introduced in 2.05.74.

This leads to a problem for me in homebridge-hue:

`Cannot add a bridged Accessory with the same UUID ...`

The only way to work around this for me was to delete it in phoscon and restart homebridge. No other negative effect so far. But the "light" returns from time to time

realwax, 2020-02-28:
Yes, I dared to delete it, luckily without harm.

ebaauw, 2020-02-28:
> This leads to a problem for me in homebridge-hue:

See https://github.com/ebaauw/homebridge-hue/issues/635.  This issue is caused by #2490, which unfortunately was discovered (and solved) just after v2.05.74 was released.  As a workaround, you can blacklist the `/lights` resource (see https://github.com/ebaauw/homebridge-hue#configuration).  
The next release of homebridge-hue will workaround #2490.

> Yes, I dared to delete it, luckily without harm.
> But the "light" returns from time to time

You can delete the `/lights` resource for the coordinator without harm, but it will likely be re-created the next time you search for new devices and/or the next time deCONZ starts.

realwax, 2020-02-28:
@ebaauw Thank you. I will close it as it is fixed and to be released.

closed 2020-02-28 by realwax