dresden-elektronik/deconz-rest-plugin #2501: Switch is missing from deconz ui

Knapoc, 2020-02-27:
My Aqara round smart switch is showing up in phoscon and via the rest api, but is missing from deconz ui. The switch/sensor is missing from the node list and is not visible in the node map.
However it is fully functioning when I'm listing to its button events in home assistant via the deconz integration.

phoscon:
[image]

rest-api:
**GET**: `http://ip:8080/api/apikey/sensors/12`
```
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "f23b18aa5609e765feb9d01198acc080",
  "manufacturername": "LUMI",
  "mode": 1,
  "modelid": "lumi.sensor_switch",
  "name": "Smart Switch Robovac",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2020-02-27T18:38:12"
  },
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:04:23:f2:45-01-0006"
}
```

Smanar, 2020-02-28:
I don't think it's possible.
Have you check using UniqueId ?

Knapoc, 2020-02-29:
what do you mean exactly? I do have the unique id as a value in the rest api. Is there something else I should check?

Mimiix, 2020-06-05:
@Knapoc is this fixed?

Knapoc, 2020-06-05:
Yeah repairing fixed the issue.

closed 2020-06-05 by Knapoc