dresden-elektronik/deconz-rest-plugin #2502: Aqara motion sensor does not report motion status correctly

rmotapar, 2020-02-27:
Hi,
I connected an aqara motion sensor successfully to the conbee ii using phoscon app. I am using hassio with deconz add-on. The motion sensor works well for the first few hours but stop detecting motion after that. When it is working well, I see two events coming from the sensor as soon as there is some motion. One is LUX and another is motion status. But after a while, it reports only LUX value when there is motion but does not update the motion status. This is the same in home assistant UI and also in phoscon app. The sensor does start reporting motion status after I move it around a bit or press the button on the side.

Conbee is connected via a USB 2.0 extension cable. The fact that it is reporting LUX correctly as soon as there is motion tells me that there are no connection/signal strength issues. Any idea what might be happening here?

rmotapar, 2020-02-28:
just bumping this to see if anyone has any suggestions on how to fix this issue. Could this be because of lack of any repeaters in my setup? I have a hubitat setup (trying to move from hubitat to homeassistant/deconz) and I connected the same Aqara sensor to hubitat yesterday evening using a custom device handler and it did not fail yet - I am seeing both LUX and motion status updates instantaneously. I will continue testing this with hubitat and will report back if I see the same issue where it reports LUX but not motion status. 

Would really appreciate any help you can provide in debugging the issue with deconz. Thanks!

SwoopX, 2020-03-03:
If I'm not mistaken, it takes like 2 minutes for the sensor to change from occupied to unoccupied. Have you waited that long? Or is it the change from nomotion to motion? Eventually, that didn't play a role when you first started using it.

rmotapar, 2020-03-03:
It's a change from no-motion to motion. The sensor was not sending any motion events unless I reset it but it does send LUX as soon as there is motion (quite strange). I ended up returning the sensor back to seller so we can close this issue.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

Sinthos, 2020-07-02:
I have the same problem since a couple of days. I am using Phoscon and the RapsBee 1. The Aqara sensors have been working fine for several months and now they only report Lux levels and occasionaly movements.I have placed the sensors right next to the gateway, changed batteries, reset and changed the channel. When I press the ping button on the Aqara sensors, they indicate the movement once and then it is detected in maybe 10% of cases. Although Lux is always played back without problems. 

Firmware 26350500
Version 2.05.78 / 22.5.2020When I press the ping button on the Aqara sensors, they indicate the movement once and then it is detected in maybe 10% of cases. Although Lux is always played back without problems.

When I press the ping button on the Aqara sensors, they indicate the movement once and then it is detected in maybe 10% of cases. Although Lux is always reported without problems. Sometimes the sensor reports the movement with a delay of a few minutes.

Firmware 26350500
Version 2.05.78 / 22.5.2020

Mimiix, 2020-07-02:
@Sinthos Hijacking a issue isn't appreciated, if you want you can open another [bug report](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Bug+report&template=bug_report.md&title=)