dresden-elektronik/deconz-rest-plugin #2503: deCONZ doesn't pick up NWK group multicast message

ebaauw, 2020-02-27:
I'm adding support for the Bitron Remote (#2392).  Fairly straightforward: bind the client clusters to a group (0x0006).  Add a (Hue) light to the group.  And indeed, the remote controls the light.  However, deCONZ doesn't receive the group commands.

Looking at the sniffer, the Bitron sends a package I haven't seen before.  It seems to address the group at NWK level, instead of at APS level.  The Hue light is perfectly happy accepting this as a group command: after removing the group from it, it no longer reacts to the remote.  
As a workaround, I added a unicast binding from the remote to the coordinator; after that, the REST API plugin generates the button events alright.

I think this issue originates in the RaspBee firmware (0x26340500) and/or deCONZ core (2.05.74).  I don't even get an APS-DATA.indication in the deCONZ log for the multicast messages.  @manup is this something that can be fixed?

The dimmer is fairly old (date code 20150518; ZCL version 1), with a TI mac address and a 0x0000 manufacturer code.  Is this a feature that's been retired from the standard?
```
ZigBee Encapsulation Protocol, Channel: 11, Length: 53
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0xcad5
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 68
    Destination PAN: 0x476f
    Destination: 0xffff
    Source: 0xcad5
    [Extended Source: MaxStrea_00:41:83:1c:ae (00:13:a2:00:41:83:1c:ae)]
    [Origin: 3]
    TI CC24xx-format metadata: FCS OK
ZigBee Network Layer Data, Dst: 0x0006, Src: 0xf893
    Frame Control Field: 0x0348, Frame Type: Data, Discover Route: Enable, Multicast, Security Data
        .... .... .... ..00 = Frame Type: Data (0x0)
        .... .... ..00 10.. = Protocol Version: 2
        .... .... 01.. .... = Discover Route: Enable (0x1)
        .... ...1 .... .... = Multicast: True
        .... ..1. .... .... = Security: True
        .... .0.. .... .... = Source Route: False
        .... 0... .... .... = Destination: False
        ...0 .... .... .... = Extended Source: False
        ..0. .... .... .... = End Device Initiator: False
    Destination: 0x0006
    Source: 0xf893
    Radius: 28
    Sequence Number: 203
    [Extended Source: TexasIns_00:09:55:b0:6f (00:12:4b:00:09:55:b0:6f)]
    [Origin: 40]
    Multicast Control Field: 0x45
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 255, Src Endpt: 1
    Frame Control Field: Data (0x08)
        .... ..00 = Frame Type: Data (0x0)
        .... 10.. = Delivery Mode: Broadcast (0x2)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 255
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 108
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 1
    Command: Step with OnOff (0x06)
    Payload
```

Hypfer, 2020-04-24:
I'm not seeing any button events with this remote which is probably the issue described here.

>As a workaround, I added a unicast binding from the remote to the coordinator

How do you do that?

ebaauw, 2020-04-24:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/b3086c6009154aa1b5cdb89febb5f177912952e9/de_web_plugin.cpp#L14537-L14578