dresden-elektronik/deconz-rest-plugin #2505: Phoscon Tradfri Remote issues

xxLeoxx93, 2020-02-27:
Hello,

I've paired my Tradtri Remote with my Conbee 2 stick but it behaves very strange:
When I press the middle button it turns of almost all lights in the building. It's only assigned to one group with 3 switches and nothing else.
Also the colour temp modification isn't working, only dimming and on/off.

I also have a misterious "Group 0" in my group list that I a) never created, b) can't delete and c) where I can't remove any devices. Maybe that's connected to this issue.

I also see in my domoticz addin the group "GROUP_TRADFRI_on/off_switch_" reported that I don't see in phoscon at all (and it never existed there).

Gateway firmware: 264A0700
Version: 2.05.74
Product: Tradtri Remote
Product Version: 2.3.014

Thanks for any help!

SwoopX, 2020-02-27:
#2485 chime in there

xxLeoxx93, 2020-03-01:
@SwoopX  Thanks for the hint, that explains the issue with the remote, but is that also related to the "Group 0" that I've and can't delete?

Smanar, 2020-03-01:
You can have some group created by API but invisible in phoscon.

The group 0 is a group for "all bulbs" but idk why it's hidden on some configuration and not other https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2065

The group "GROUP_TRADFRI_on/off_switch" is created when you add the ikea remote, it's a special group, Xiaomi don't make one.

This group is for "direct connexion" all device in this group can work without the conbee, but color can't work.

There is too a normal method to associate a switch to a bulb on Phosocn without using this group.

xxLeoxx93, 2020-03-01:
Thanks for your help! That means for me I need to keep an eye on #2485 and #2065 . Thanks again!

closed 2020-03-01 by xxLeoxx93