dresden-elektronik/deconz-rest-plugin #2509: Conbee 2 not functional after manual firmware update failure

jacgl, 2020-02-28:
Hi,
I am using Conbee 2 with ubuntu 18.04. Today I started manual FW upgrade with last image (264A0700->26530700), according to ubuntu procedure.
Unfortunately, GCFFlasher_internal stopped the update in about 30% of progress with exit message like lost connection to device (reason? really not known for me).
Now Conbee dongle is not functional, it periodically appears on usb as /dev/ttyACM0 for a second or 2, then dissapears for a few seconds. Anyway, it is not visible by deconz/phoscon.
What can be the next step to investigate?
Regards

Smanar, 2020-02-28:
Perhaps that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493 ?

jacgl, 2020-02-28:
Thanks, it works.
But I had to start the command many, many times. At the and FW programmed correctly.

salopette, 2020-02-29:
Are there any new FW for the Conbee-ll?

I didn't find it in the wiky.

ConBee II
wget http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26490700.bin.GCF

[image]

Knapoc, 2020-02-29:
> Are there any new FW for the Conbee-ll?

The newest firmware release for Conbee II is `0x264a0700` as [found here](http://deconz.dresden-elektronik.de/deconz-firmware/).

ebaauw, 2020-02-29:
> it seems 0x26530700.bin.GCF is the newest one

Beware, that's a beta version, and doesn't work in all situations, see #1261.  I think `4a` is the latest stable release.  It's distributed with recent versions of deCONZ in installed to `/usr/share/deCONZ/firmware`.

Knapoc, 2020-02-29:
I just recognised that and will update my previous post to avoid frustration :).

salopette, 2020-02-29:
ok, but it is really sad that Wiky is not kept up to date.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix