dresden-elektronik/deconz-rest-plugin #2511: REST-api reports wrong position/brightness value for Fyrtur

Knapoc, 2020-02-29:
In the releases after 2.05.71 there's a mismatch between the value reported from deconz GUI and the rest-api.

As far as I understand the rest api should report `bri: 255` and not `bri: 254` at a lift percentage of 100

REST output:
```json
{
  "etag": "5481e5d77f899bdb4c2cb42d2cfca47a",
  "hascolor": false,
  "manufacturername": "IKEA of Sweden",
  "modelid": "FYRTUR block-out roller blind",
  "name": "Fyrtur 1",
  "state": {
    "alert": "none",
    "bri": 254,
    "on": true,
    "reachable": true
  },
  "swversion": "2.2.009",
  "type": "Window covering device",
  "uniqueid": "cc:cc:cc:ff:fe:f4:c9:24-01"
}
```

[image]

Kane610, 2020-02-29:
From what I've seen in comments from @ebaauw 254 is indeed the correct value

ebaauw, 2020-02-29:
Correct.  254 is the max value for `state.bri`.

Knapoc, 2020-02-29:
Ah I just saw the pull request. It seems the [documentation](https://dresden-elektronik.github.io/deconz-rest-doc/lights/#getstate) is outdated or I'm missing something in my understanding: 

Field | Type | Description
-- | -- | --
state.bri | Number (0..255) | Brightness of the light. Depending on the light type 0 might not mean visible "off" but minimum brightness.

But it seems that 254 is the appropriate value, since lights show the same on full brightness.

closed 2020-02-29 by Knapoc