dresden-elektronik/deconz-rest-plugin #2512: Remote access to deConz GUI

Pedder007, 2020-02-29:
New posting, as I occasionally posted the following first in an already closed issue:

Hi, I've got a propper running Raspi 3B with a deConz II Stick since roughly a year. It connects my Pimatic HA with the ZigBee world (Hue bulbs, a couple of Xiaomi sensors and also some Tradfri devices). All works more or less propper and stable. Up to know I was always able to bind the devices by the Phoscon App and discover them afterwords in my HA.
As a further enhancement, I currently bought a light- and motion/light-sensors (Xiaomi GZCGQ01LM and Phil Hui motion indoor), but the setup of both now drives me a bit crazy since some days.
What I alredy found is, that there are still existing threads (dresden-elektronik #1559  and #2380 ) which show, that they are both tricky to connect and that some parameters have to be configured manually via the deConz Gui (not possible via the Phoscon App!).

Now my question, as I'm running deConz in headless mode on the Raspi:
Is it possible to access the deConz GUI, e. g. from a PC having installed the deConz App on it, without changing anything in my Raspi/deConz configuration?
As I'm not that deap in the technics, I fear a little bit to crash my HA setup by switching the mode of deConz from headless to 'incl. GUI' mode. I found another thread (#2045  ) with a similiar question, but it seems not have been answered in terms of a 'simple' solution, which the usage of docker definitely would not be for me.

Happy for every hint to tackle this 😄
Hopeful regards Pedder

ebaauw, 2020-02-29:
The GUI is integral part of the deCONZ programme; it cannot be detached like the Phoscon web app, that uses the REST API to connect to deCONZ.  It's by far easiest to run deCONZ with the GUI enabled all the time, even on a headless system.  Use screen sharing (VNC) to access the graphical environment remotely.

If you're really experienced with X11, you could try and use the `DISPLAY` variable to make deCONZ display the GUI on a remote X server (screen).  Not sure I'd recommend this; deCONZ will likely crash when the X server is not available.

Pedder007, 2020-02-29:
OK thanks @ebaauw!
Then I will give it a try to activate the GUI on the Raspi to tackle my sensor issues.
... and pray to god that my HA setup will not crash.
Thankful regards
Pedder

sub0ne, 2020-03-01:
> It's by far easiest to run deCONZ with the GUI enabled all the time, even on a headless system.

meaning installing deconz-gui instead of deconz package? How can I switch from deconz to deconz-gui in a productive environment?

ebaauw, 2020-03-01:
There is only one deconz package, and only one deconz program.  The display of the GUI might be suppressed using a command-line option when starting the program.  That’s what the deconz service does, where the deconz-gui service starts the program without this option.

So you want to boot your headless (no monitor, keyboard, nor mouse) Pi into the graphical environment, which can be accessed over VNC, and you want to enable the deconz-gui service and disable the deconz service.

Note that deconz crashes when it tries to display the GUI when the graphical environment hasn’t started.  This happens on boot, where the deconz-gui service is started too early.  See #2482 for a fix.

Pedder007, 2020-03-06:
Hi again, I now followed the steps in #1405, but still wasn't able to start the gui.
I checked the status by sudo systemctl status deconz-gui and it seems to run, as it echos:
```
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-03-06 21:22:48 CET; 1min 26s ago
 Main PID: 18393 (deCONZ)
    Tasks: 12 (limit: 4915)
   CGroup: /system.slice/deconz-gui.service
           ├─18393 /usr/bin/deCONZ --http-port=80
           ├─18517 dbus-launch --autolaunch d8c7af110d2f4619b3b48e8653b1b407 --binary-syntax --close
           └─18518 /usr/bin/dbus-daemon --fork --print-pid 5 --print-address 7 --session

Mär 06 21:22:48 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Mär 06 21:22:48 raspberrypi deCONZ[18393]: libEGL warning: DRI2: failed to authenticate
Mär 06 21:22:49 raspberrypi deCONZ[18393]: libpng warning: iCCP: known incorrect sRGB profile
Mär 06 21:23:49 raspberrypi systemd[1]: [/lib/systemd/system/deconz-gui.service:11] Unknown lvalue '
```
What else may have I done wrong?
I here and there read something about differently used connections to headless Raspis.
I'm using XRDP. Might that be the problem?

Pedder007, 2020-03-06:
Continued testing:
Found #2253: stopped both, deconz and deconz-gui
Then started the GUI via gksudo and suddenly the deConz App started in my XRDP window.
The start page showed in the access menue Raspberry and ConBee II.
I tried both, but besides the status 'connecting to device' in the down left corner nothing happened.
I then just gave it a try to open in parallel the Phoscon App in my PC browser which worked.
... and now it became cruical:
Suddenly my Hue Motion Sensor started showing the illumination values!?
I have no clue why, but that was one reason why I tried to access the stick by the deConz App.

However,
if one has an idea why the connection might have failed, I would be happy to let me know.
If there is a chance to connect, I could then also try to connect my Xiaomi Light Sensor #2300 :-)

Pedder007, 2020-03-30:
Hi again, I will try again to get some help :-)
As the above mentioned approach seems not have been a stable solution (meanwhile two hue motion sensors are sending their data only with large occasional interrupts) please read following what I tried:
1. Started a XRDP session from my PC on the Raspbee.
2. Stopped my house automation by 'sudo service pimatic stop'
3. Stopped deCONZ by 'sudo service deconz stop'
4. Started deCONZ GUI-Version by  'sudo service deconz-gui start'
5. called gksudo
6. typed 'deCONZ' into the command line by using user 'root'
7. deCONZ GUI started and offered to connect by 'Raspbee' or by 'ConBee II'
8. Both doesn't work. The status line in the lower left corner shows 'connecting network' but doesn't join the network in the end.
Someone an idea what I could try else?
Many thanks in advance.

ebaauw, 2020-03-30:
You can only have one instance of deCONZ running.  If you already started it through the deconz-gui service, you cannot start it again manually.

Pedder007, 2020-03-30:
OMG!, ..... now I got it!!!
I thought the whole time, that by starting the deCONZ App on the Raspbee, that is also only a kind of frontend..., I only had to skip bullet No 4, in my above mentioned thread...
Thanks again, sometimes it takes a while to interprete the things in the right way ... :-)

I will no do some addtional tests in the eveing. What I just quickly saw, is that one of the hue motions has a bad connectivity (orange wire) and that the router has > 15 other devices on it. So I may have to re-arange the physical distribution of the routers ...
Thanks again :-)

Pedder007, 2020-04-02:
So in the end I found my way how to start the deCONZ GUI and maintain the details in single device configurations.
Now I‘m fighting with the same issues, with regard to the Hue Motion Sensors (esp. light measurement), like multiple other users here on Github.
I will close here and jump across on to already existing threads for respective questions 😊👍

closed 2020-04-02 by Pedder007