dresden-elektronik/deconz-rest-plugin #2513: Installed Deconz worked fine! Next day lights do not respond.

skitzow88, 2020-02-29:
I installed Deconz and ordered a Conbee2 to replace my hue and aqara hub. Everything worked fine when installing only thing i didnt get to work is to add independent bulbs within phoscon. I worked arround this by adding light groups. The next morning my lights wont respond. (Both aqara and hue bulbs and switches stopped working, sensors are working fine. When running Homebridge -I -D i dont see any errors.

Smanar, 2020-02-29:
So you have loose bulbs and not sensors ?
Thery are coming back if you make a power on/off ?

Are you using the last firmware ?

skitzow88, 2020-02-29:
Yes i lost all the bulbs (they are greyed out within the phoscon webapp) Sensors working fine. 
No they wont come back after rebooting/shutting down. 

My gateway information: 

[image]

The greyed out bulbs: 
[image]

Sensors: 

[image]

Smanar, 2020-02-29:
Have you try to take a look on last posts on this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

On my side I m thinking like you, not an hardware or connexion problem (but sensors need more time to be marked as unrecheable, are you sure they are working ATM ?)

skitzow88, 2020-02-29:
Well i did read the article, but thats related to losing some light bulbs, my setup keeps losing every light bulb possible. At this moment i can also not confirm if the sensor respond. I will start a fresh reinstall of  homebridge, conbee and all of the plugins. Will report back when everything is reinstalled.

Smanar, 2020-02-29:
I don't thnik re-install will change something, as deconz is working as stand alone app.

skitzow88, 2020-02-29:
i agree, but will be easier to troubleshoot when everything is freshly setup. Almost done, got homebridge up and running again. Still gotta install 2 plugins and then deconz

skitzow88, 2020-02-29:
Well reinstalling everything did gave me the option to expose the lightbulbs separately to homebridge this time! At the moment everything is running as expected.

[image]

[image]

[image]

skitzow88, 2020-03-01:
Well next day everything is still working as expected! HomeKit automations with smart switches don't respond as they should but i think homekit software is to blame. When i configure the smart switches within in phoscon they work fine.

skitzow88, 2020-03-01:
Issue closed, im quite sure that i did something wrong when configuring/installing Deconz the first time. A fresh installation did solve all problems i was experiencing in the first configuration. Now i can see more options within Phoscon.

closed 2020-03-01 by skitzow88