dresden-elektronik/deconz-rest-plugin #2514: The plugin no longer compiles after todays commits

nikolauzi11, 2020-02-29:
I get this error:

`g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.05.74\" -DGW_SW_DATE=1582982159 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"b0da16a651df2dd41c3d2c949637300eef65b1d2\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26330500 -DGW_MIN_R21_FW_VERSION=0x26490700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/de_web_widget.o de_web_widget.cpp
de_web_plugin.cpp: In member function ?void DeRestPluginPrivate::addSensorNode(const deCONZ::Node*, const SensorFingerprint&, const QString&, const QString&, const QString&)?:
de_web_plugin.cpp:5215:14: error: ?sensor? was not declared in this scope
             !sensor.type().endsWith(QLatin1String("Battery")))
              ^~~~~~
de_web_plugin.cpp:5215:14: note: suggested alternative: ?sensors?
             !sensor.type().endsWith(QLatin1String("Battery")))
              ^~~~~~
              sensors
Makefile.Release:635: recipe for target 'release/de_web_plugin.o' failed
make[1]: *** [release/de_web_plugin.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/uscholz/deconz-rest-plugin'
Makefile:40: recipe for target 'release' failed
make: *** [release] Error 2`

ebaauw, 2020-02-29:
Same here.  Faulty commit is https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ffcda5e66ac7bc5700cf91cf460723c28941f55f.

nikolauzi11, 2020-02-29:
Works again. Thanks

closed 2020-02-29 by nikolauzi11