dresden-elektronik/deconz-rest-plugin #2516: 26530700 not working

sub0ne, 2020-03-01:
Just did an firmware upgrade from 264A0700 to 26530700. Update did run well and after a reboot all my lights and sensore showed up in deconz (2.05.74). Sadly none of my bulbs turned on when I activated them in deconz. So I switched back to 264A0700 and everything worked fine again.

I've read that 26530700 is beta, but is this a known issue with 26530700? Is it normal that my lights won't work after firmware update? Do they have to be repaired?

ebaauw, 2020-03-01:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-593041256

danielkihlgren, 2020-03-01:
Yes, it is known. 
I have the same problem.

sub0ne, 2020-03-02:
for me not only some light weren't working, none of my lights did work. I couldn't even activate them by rest api through iobroker/homebridge.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix