dresden-elektronik/deconz-rest-plugin #2517: Cannot access gateway through Phoscon app

Sjonnie2018, 2020-03-01:
I know the Phoscon app is closed source and not supported by the community but I hope you have experienced something similar and are able to help me.

Since today I cannot access the gateway through the Phoscon app. It says:

[image]

I have reset the device and unplugged it to see if I could connect again but unfortunately not.

Deconz and deconz-gui are working fine as are the devices and scenes. It is just accessing the gateway that doesn't work.

I would like to be able to access the gateway before I update the firmware that was released on February 24th. Although it is said to be beta so I might wait till final release. Even so, I still would like to be able to access the gateway through the app.

Smanar suggested on the Domoticz forum to do a

`ps ax | grep deCONZ`

which resulted in this

`pi@RPi3:~ $ ps ax | grep deCONZ
  370 ?        Ss    24:14 /bin/bash /usr/bin/deCONZ-update2.sh
  753 pts/0    S+     0:00 grep --color=auto deCONZ
 3980 ?        Ssl   32:01 /usr/bin/deCONZ -platform minimal --http-port=80
11691 ?        Ss    36:05 /bin/bash /usr/bin/deCONZ-WIFI2.sh`

I would hate to reset the device. I have 23 nodes and a fully configured Domoticz environment. Not to mention a rather extensive Dashticz dashboard. So not looking forward to replace all IDX's with new ones 

Any tips?

TIA

Greetz,

Sjonnie

PS. I read the suggested posts #1070, #2097, #490 and #304 but they do not correspond to my problem.

cs42, 2020-03-01:
I had this problem too since yesterday. I had installed 2.05.74 from the beta repository (running deconz on an NUC with ubuntu). Tried many things, but for me the solution was to install 2.05.74 from the **stable** repository! Looks like a discovery problem, as sometimes I got redirects to phoscon.de for login.

BTW, is it possible to disable this _global_ discovery using phoscon.de? I do not need this...

Smanar, 2020-03-01:
If you use only the local url, you don't use phoscon discovery ?

Sjonnie2018, 2020-03-01:
@cs42 That did the trick! Thank you very much for your sift reply and solution. Much appreciated!

Sjonnie2018, 2020-03-01:
I am going to close this issue because it is solved and there are way too many open issues on this "board".

closed 2020-03-01 by Sjonnie2018