dresden-elektronik/deconz-rest-plugin #2518: Ikea Tradfri Firmware 2.3.x E17 not usable with deconz

realwax, 2020-03-01:
The experience with a mesh of 20 E14 bulbs upgrade to 2.3.x is bad.
Issues:
Soft on gone (fade in at switch on) 
Recalling scenes freezes deconz on and off and other recall commands in that group for 10 to 15 seconds before able to send another scene recall or switch off. 
Lights start blinking (unknown if firmware issue or deconz reaction.) bulb does not respond.

Though there some improvements in terms of usability it not recommended from my experience. 

Please use this thread as discussion. Maybe deconz can be improved, maybe simple do not go to fw 2.x on your bulbs. 

With a downgrade to 1. 2.x on e14 everything is fine and tradfri operation was improved in this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

Lemocin, 2020-03-28:
I have the white spectrum bulbs.
Switching on with scenes works for me, same as other brightness/color, but the bulb doesn't let me turn it off. Sometimes in flashes in random colors and sometimes it doesn't do nothing at all, except hanging, but i didn't need to powercycle after 10-20 seconds, i can control them again. 
After the downgrade it's working again.

realwax, 2020-03-29:
@Lemocin Thanks for your contribution. We need to get more attention here. @tubalainen @ebaauw @djwlindenaar @manup

ebaauw, 2020-03-29:
There is a bug in most/all (?) versions of the firmware of most/all (?) IKEA Trådfri lights, that it won't accept new commands while it's still executing a transition.  So e.g. a PUT of `{"bri": 254, "transitiontime": 600}` will render the light unresponsive to new commands for a minute.  See #1030 and the other issues mentioned there.

This seems to be a similar firmware bug.  What `transitiontime` do you use recalling the scene?  How did you create the scene?  Or using _Store Scene_ or _Add Scene_/_Enhanced Add Scene_?  The latter doesn't seem to work for IKEA, see #2507.  Do you seem the same behaviour when the light is connected to a Hue bridge or Trådfri hub?

tubalainen, 2020-03-29:
I am planning to add smoke detectors to my home and I found that all of the ones that I am willing to buy (based on functionality and reliability) all working so-so in deconz (some with issues more than 2-3 years old).

Lets hope that we soon can get the FW and Deconz issues with mesh routing sorted so I personally dont have to "pull the trigger" and move away from this otherwise fantastic software!

realwax, 2020-03-30:
@ebaauw 
In both cases of Ikea FW 1.2.x and 2.3.x the operation is the following with different results. (So from my perspective the so cold bug you were referencing seems to be the case in later versions) 

1) Using Phoscon setting lights in a group to desired settings and storing the scene or using the deconz plugin in iobroker doing the same - a deconz api store scene.
2) recalling the the scene and here is the difference.
 
Version 1.2.x - scene will be recalled. Regardless the transition time of 3-10 seconds, you can recall another scene while in active transition, or turn lights off. Commands will be executed.

Version 2.3.x - scene will be recalled. Regardless of the transition time (in active transition or after transition) you have to wait a couple of seconds more that deconz can handle lights again or they respond. around 10-15 seconds. If you try to recall a scene before that timeframe or turn off the lights some respond others not and the group seems to be frozen for some more time.

Mimiix, 2020-06-05:
@realwax is this still a issue?

realwax, 2020-06-05:
@Mimiix Currently I can't tell as I had my bulbs flashed to 1.2.x for smooth operation. I am currently trying to test again. I start flashing my E14s to 117C-2201-23050631.zigbee which is version 2.3.050. I will come back with my usability experience in 2-3 days as flashing takes some time and manual control is required....

realwax, 2020-06-05:
Update - it seems like improvements are working...!
A group of 3 and another containing 4 bulbs is working! It seems a little slower than before in terms of while in transition accepting another scene switch stopping the current one, or being turned off while in transition but the major stoppers seem to be gone. I will do more testing after all my bulbs shrare the same version.

Mimiix, 2020-06-05:
Sounds good!

realwax, 2020-06-06:
@Mimiix 
SUCCESS! 

As you have confirmed the OTAU bug recently, you can imagine flashing takes some time and need plenty restarts. 
BUT nevertheless I coming back with good news. I tested two groups of my bulbs that have been updated back to 2.3.0.50 successfully. I am still in the process of updating and testing. But to be fair - the topic's /issue's name is not right anymore because bulbs and groups and scenes are working. 
[image]
In terms of group command even better with the bottom line changing CT in phoscon gui sometimes needs more than one command. But that has been discussed with @ebaauw in another thread where he stated commands are split into several messages and it might be where "something" happens. Anyhow - for this kind of issue we can raise new items. 

This is closed.
Thank you for your support and work!
Kind regards from Vienna :) 
Wolfgang

closed 2020-06-06 by realwax

realwax, 2020-06-06:
additional note: (Can't really state on which side the issues have been mitigated. Guess on both)
using latest ikea fw 2.3.050 release at 20.5.2020

Release Version: 1.12.1
20th May 2020
New features and changes in Accessories:
WS 1.0 (E14, E27, GU10) V-2.3.050.
Fixed On Off State after OTA Upgrade.

realwax, 2020-06-06:
@Mimiix One issue is still left - find it here https://github.com/dresden-elektronik/phoscon-app-beta/issues/52 (it is api related not phoscon gui) you can reproduce it with simple api calls.

Mimiix, 2020-06-07:
Can you create a new issue for that?

realwax, 2020-06-07:
> Can you create a new issue for that?

sure!