dresden-elektronik/deconz-rest-plugin #2525: Coordinator gets name of new device (Elvis Container)

lukicsl, 2020-03-02:
I upgraded my docker container to the newest version Elvis
I tried to add a new GU10 Ikea lamp.
After adding the lamp through Phoscon and renaming it I see that the coordinator got the new name of the lamp.

[image]

The light has also no attributes in phoscon:
[image]

manup, 2020-03-02:
An update will be released shortly, which filters the Gateway resources out of the lights list in the Phoscon App.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix