dresden-elektronik/deconz-rest-plugin #2526: Ikea E27 WS - not reporting color_temp value

miki9987, 2020-03-02:
Issue still not fixed. Details here..
https://github.com/home-assistant/home-assistant/issues/26424

Mimiix, 2020-06-05:
Is this still a issue?

xaviml, 2020-06-05:
Hi @Mimiix,

I can confirm this issue is still open. I set up a "TRADFRI bulb E27 WS opal 1000lm" light with deconz  (2.05.75, through Home Assistant addons) 2 days ago and I confirm that deconz doesn't report the "color_temp" to Home Assistant. This issue is related to:
- https://github.com/home-assistant/core/issues/26424 (closed because of duplication)
- https://github.com/home-assistant/core/issues/24143 (closed by mistake, people still reporting the issue after the fix)
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1861 (still open)
- https://github.com/xaviml/controllerx/issues/56 (reported on my project)

In fact, this issue is a blocker for users using this light with deconz and ControllerX, since ControllerX needs that attribute to change the color temperature.

If there is any information on my side you want me to provide, just let me know.

Cheers,
Xavi M.

ebaauw, 2020-06-05:
Fix is in .76 / .77.

EarlOfBurton, 2020-06-20:
Hello @ebaauw  - excuse me, I'm rather new to github.
I'm using home assistant with deconz and IKEA TRADFI bulb E27 WS opal 1000lm (V2.0.022) and I cannot find out how to update this rest-api so that the ikea Tradfi Bulb Color temperature is working.
Could you tell me how I can update the deconz rest-api in home hassistant (it does not show up in any update-list)? I looked through the internet for one hour but no luck for me so far.

ebaauw, 2020-06-20:
No, I cannot not.  I don't use Home Assistant.

EarlOfBurton, 2020-06-20:
OK, thanks @ebaauw for the quick response.
Anybody else knows if the bug is still in the home assistant add-on or how to get a manual update for that?
Kind regards

xydix, 2020-07-10:
I have a IKEA Trådfri GU10.
I can choose the coldest or the warmest colortemp. Nothing between. This is from Home Assistant.
Haven't tried from deconz yet. (The bulb isn't in a light group)
Bulb firmware 2.0.022
Deconz 2.05.78
Home assistant 0.112.3

xaviml, 2020-07-11:
Hi all,

Related to what @xydix said, now the color temperature works if you use callnservicr and send a number between 153 and 500 (min and max). However, rn lovelace the min-max is 153-65535, so when you move the slider to min is coolest but if you move it a little bit to the right the value is already higher than 500 and it turns warm. This is what is happening. I have not checked the code, but the problem is that the max mired is set to 65535 instead of 500.

Cheers,
Xavi M.

xydix, 2020-07-15:
I can confirm this.
If i try to set color_temp: 370 for example I can get a mix och warm and cold light.
Is this a bug in the deconz integration in Home Assistant or is it a bug in deconz?

EarlOfBurton, 2020-07-23:
I can confirm what @xydix and @xaviml say. my E27 1000lm IKEA TRADFI lamp has a working range from 153-454 although the range in lovelace lets me set from 153-65535.
When I set it to "coldest" temperature initially it jumps from 153 to 250. After that I can go to 153. Not sure if the lowest value for the lamp is 153 or 250. Thanks for the great software btw - looking forward to integrating more IKEA equipment fully when everything is working properly.

ebaauw, 2020-07-24:
About colour temperature ranges:
- Zigbee uses a uint16 to store/set the mired value, which theoretically supports values 0 to 65535 (or 0x0000 to 0xFFFF)
- Zigbee defines valid mired values for colour temperature in the range 1 to 65279 (or 0x0001, ..., 0xFEFF).  That corresponds to 15, ..., 1,000,000 Kelvin, which obviously no light can physically reproduce.
- I think a light could theoretically reproduce colour temperature from ~1500 to ~7000 Kelvin, which roughly corresponds to a mired range from ~140 to ~650.  Which temperatures the light can physically reproduce depends on the types and number of LEDs it uses to mix the colour temperature.  This can be further limited by the light firmware, which needs to translate the mired value in the brightness of the various LEDs.
- Zigbee defines attributes for a light to report what range of colour temperatures it supports, but not all lights (or firmware versions) support this.  Worse: some lights report bogus values for these.  I think the ZLL firmware of the Trådfri suffered from this, but the ZB3 firmware shows the correct values.  Note that for the API to pick up these values you need to re-read the attributes in the _Color Control_ clusters in the GUI, after the firmware update.  Alternatively, re-pair the light.
- Trådfri lights typically support 250 to 454 (2200 to 4000 Kelvin) , Hue lights 153 to 500 (2000 to 6500 Kelvin), OSRAM 153 to 370 (2700 to 6500 Kelvin).
- Note the mired is inverted: higher mired values correspond to lower (warmer, more yellow-ish) temperatures, lower mired values correspond to higher (cooler, more blue-ish) temperatures.
-

johnekarlsson, 2020-08-02:
I also have this problem. Because of it I have decided to change most lamps to hue...

EarlOfBurton, 2020-08-04:
Wow @ebaauw, you're very knowledgeable and smart.
Would it be possible to a) redefine the scale in the hassio deconz-plugin at some point accordingly or b) use color temp from 250-454 so that the slider range is somehow to use in practical life?
Practically the is no difference in the slider positions for 'as cold as possible' (250) and 'as warm as possible' (454), see Screenshots attached.
[image]
[image]

SwoopX, 2020-08-04:
This is something which needs to be changed by the HA team. We don't have any influence on that.

ebaauw, 2020-08-04:
With the ZB3 firmware, the light should report the correct min and max values.  If you read the _Color Control_ attributes from the GUI, the API should reflect these, and API clients, incl. HA should pick them up.

EarlOfBurton, 2020-08-04:
@ebaauw, excuse my stupidity, what is the ZB3 firmware? ZigBee3 firmware? For the IKEA lamps?
I tried to do a OTA update with the vnc connection from home assistant to deconz, but I could not get the update starting, it always was at "queued" state. Now they say "no file" although there are the files in the folder. There is a "plain" 117C22010631.zigbee file and a  1035514-2.1TRADFI-bulb-ws-2.3.007.ota.ota.signed file. Which one to use? And what does the "2.1" in the filename stand for? I see on other tradfi firmware that there are also "5.1", "2.2", "3.3" - what do they stand for?
At the moment the 1000lm cct e27 lamps says vendor ßx117c, image 0x4204, version 0x20029623 Thanks for your help - I'm a little confused right now.

ebaauw, 2020-08-04:
> what is the ZB3 firmware? ZigBee3 firmware?

Yes.

> Now they say "no file" although there are the files in the folder.

So it has probably updated overnight.  You can check by reading the _Basic_ cluster attributes: Zigbee 3 devices have a _ZCL Version_ of 3.  Note that the GUI doesn't refresh automatically after a firmware update; you need to re-read the descriptors for the GUI to show HA instead of ZLL.

> There is a "plain" 117C22010631.zigbee file and a 1035514-2.1TRADFI-bulb-ws-2.3.007.ota.ota.signed file.

See https://github.com/dresden-elektronik/deconz-ota-plugin: the OTAU plugin copies the files on startup.  I think you miscopied the filename; it should be `117C-2201-23050631.zigbee`.

> And what does the "2.1" in the filename stand for? I see on other tradfi firmware that there are also "5.1", "2.2", "3.3" - what do they stand for?

You'll have to ask IKEA.

> At the moment the 1000lm cct e27 lamps says vendor ßx117c, image 0x4204, version 0x20029623 

I think that's the latest version (or at least the latest version I've downloaded, but I don't check very often).

But did you read the _Color Control_ attributes?

EarlOfBurton, 2020-08-04:
Thanks a lot, @ebaauw! I'm making progress with your help. Indeed the lamps show ZCL version 2. There is an update running now, for the first time. :) 
[image]

johnekarlsson, 2020-08-04:
Hi! I'm reading this and thing that I should probably update all my trådfri lights for the first time...

I have succeeded to update, thanks to your instructions. So now my listed firmware is the latest. However, my ZCL-version still show 1. Is this expected? Is this going to be a problem?

Thanks again for your help so far!

EarlOfBurton, 2020-08-05:
@ebaauw, it did update over night, I took power off the lamps in the morning and repowered them, I do have a color control field, but the ZCL Version still says "2" - the SW build version is updated. I re-read the information manually of course. What am I doing wrong? Thank you for your kind help.
[image]
[image]

ebaauw, 2020-08-05:
Well, the endpoint shows ZHA instead of ZLL, so that’s OK.

Did you read the min/max values from the _Color Control_ cluster?

EarlOfBurton, 2020-08-05:
The newly read Color Control cluster is as follows:
[image]
[image]

There is a 65535 that makes no sense. Should I adjust values in this table manually? Will this change the max. values in home assistant?

ebaauw, 2020-08-05:
So 0x400B and 0x400C show the correct values.  Note that these are read-only.  The should be reflected in the API resource as `ctmin` and `ctmax`, which an API client should use to scale the slider.

> There is a 65535 that makes no sense.

Yes it does.  It means previous value.  I don’t think _PowerOn Color Temperature_ is functional for IKEA lights, however.

EarlOfBurton, 2020-08-06:
Thank you so much, @ebaauw, it worked after I got the updating understood and working.
[image]

Mimiix, 2020-08-29:
As this seems fixed, i am going to close this:)

closed 2020-08-29 by Mimiix

MacJawa, 2020-08-29:
Not sure if it's fixed @Mimiix . I'm on 2.05.79 and I could use the workaround to manually read Color Cluster, but if I have to do it everytime I add new light, it does not seem to be fixed to me.
To be honest I did not checked in detail how newly added light responds, but I know I added 2 lights few days ago and both of them showed incorrect min/max color temperature values until I manually read color cluster.
It would be great if color cluster is read when new light is added.

benganellison, 2020-10-25:
Hi, 

I just got an Ikea bulb and it does not work as expected in the phoscon interface, nor through the rest API.

[image]
[image]
[image]

I'm using post man to interact with the rest-api but i get no info about the CT mode there

get http://{{url}}/api/{{key}}/lights/7

gives me this:

{
    "colorcapabilities": 8,
    "etag": "0cc4c1e7289f94857a403c55f72cba6b",
    "hascolor": true,
    "lastannounced": "2020-10-25T20:05:52Z",
    "lastseen": "2020-10-25T20:42Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E14 CWS opal 600lm",
    "name": "Hall uppe",
    "state": {
        "alert": "none",
        "bri": 5,
        "colormode": "xy",
        "effect": "none",
        "hue": 0,
        "on": true,
        "reachable": true,
        "sat": 0,
        "xy": [
            0.5208,
            0.4563
        ]
    },
    "swversion": "1.3.013",
    "type": "Color light",
    "uniqueid": "84:2e:14:ff:fe:ff:45:1a-01"
}

i can set the color temperature using this

put http://{{url}}/api/{{key}}/lights/7/state
{"effect": "none",
"ct":454}

did i miss understand something? shouldn't I be able to get the color temperature via the api? and also get the min max values?