dresden-elektronik/deconz-rest-plugin #2527: deConz UI: No connection line but data?

xxLeoxx93, 2020-03-02:
Hello,
I'm using a network with about 50 nodes so far. Most of them are connected in the UI throu some lines. Some of them however seem to have no connection at all, but they still report data according to the phoscon app. Some of the aren't working stable tho.
So is it fine that those sensors don't have any connection at all? Or do I need to readd them until they show a line?
One smoke detector for example is ~1m away from the stick and a blue light flashes in the ui whe I press a button, but it also doen't have any line.

[image]

ebaauw, 2020-03-02:
The lines do not show connections; ZigBee doesn't maintain any.  They're drawn as deCONZ polls the ZigBee routers (yellow nodes) for their neighbour tables.  It's not uncommon for end devices (grey nodes) not to show a line.  Especially for "deep sleepers" like most Xiaomi sensors.

xxLeoxx93, 2020-03-02:
Alright, but does that mean that one of my Smoke detectors isn't deepsleeping and hence showing up here?
If I understood you correctly that means this also doesn't show how the data is routed right?
Because some of the sensors are directly next to another, but they don't show any line here.

ebaauw, 2020-03-02:
> Alright, but does that mean that one of my Smoke detectors isn't deepsleeping and hence showing up here?

It just means that it showed up the router's neighbour table.

> If I understood you correctly that means this also doesn't show how the data is routed right?

Correct.

> Because some of the sensors are directly next to another, but they don't show any line here.

The lines show neighbour table entries, not physical proximity.  These tables have a limited capacity (depending on the device firmware), and might not contain all devices within reach.

xxLeoxx93, 2020-03-03:
Thanks for the clarification!

closed 2020-03-03 by xxLeoxx93

xxLeoxx93, 2020-03-03:
@ebaauw Last questuin regarding zigbee network: is it device independent if the device can work with a repeater? Like I've a few ikea bulbs and they work great, however my xiaomi door window sensors seem not to work at all with a repeater in between. Or can a repeater only repeate a limited number of devices?

ebaauw, 2020-03-03:
A repeater is just another ZigBee router with no other functionality.  Routers can forward traffic for an unlimited number of devices, as long as it can find a route to that device.

As end devices typically turn off their radio (to preserve the battery), messages for the end-device are delivered to the parent router instead.  The parent router keeps these messages in memory (for like 7 seconds), waiting for the end device to poll it (i.e. to wake up and ask if there's any messages).  This takes up resources on the parent router, so they only support a limited number of children.  How many depends on the device firmware.  To add more end devices to your network, you need to add more routers.

Normally, an end device should detect when its parent has gone missing, and find a new parent.  Xiaomi end devices do not do this, they stick to their parent like glue.  Consequently, you want to pair these on the physical location where you'll be using them (so they pick a parent in range), and you do not want to power down your ZigBee routers (do not use 20th century wall switches to power down your smart lights).

xxLeoxx93, 2020-03-03:
That makes it actualy clear why some of them don't work properly. Thank you so much!

> How many depends on the router firmware. To add more end-devices to your network, you need to add more routers.
But that should only be a problem if you as a user want to activate the enddevice, not the other way arround like motion/temperature senosrs or door window contacts, right?

> Xiaomi end-devices do not do this, they stick to their parent.
If the power drops for example and the routers go offline - will they pick the same router again when the router is available again? And do you actualy need to re-pair them to make them chose a new router or is a powercycle enough?
Thanks again for your help! I think this is information is crutial for a stable network and many have problems because of this.

ebaauw, 2020-03-03:
> But that should only be a problem if you as a user want to activate the enddevice, not the other way arround like motion/temperature senosrs or door window contacts, right?

Wrong.  The end-device needs a parent to join the network.  If it cannot find a new parent when its old parent has gone missing, it will probably leave the network (except for Xiaomi).

> If the power drops for example and the routers go offline - will they pick the same router again when the router is available again? 

They, Xioami, yes.  They, others, they'll pick a router, might be the old one, might be a new one.  Unless the power outage took too long and they decided to leave the network.

> And do you actualy need to re-pair them to make them chose a new router or is a powercycle enough?

They should do automatically, but some don't.  In my case: the Eurotronic Spirit TRVs and the IKEA FYRTUR blinds.  Usually a power cycle (disconnect battery for a couple of seconds) is enough to bring them back.  On occasion, I also need to open the network when power cycling them.  No need to delete the resource(s) and no need to do a factory-reset.