dresden-elektronik/deconz-rest-plugin #2528: Aqara/Xiaomi Multi Sensor

flopp999, 2020-03-02:
I have a few Aqara Multi Sensor that measures Temperture/Humidity and Pressure.
In the Phoscon GUI, why does it shows 2 decimals for Humidity and not for Temperature. For me I would like to have 1 decimal on Temperature instead of 2 decimals for Humidity.
Maybe one decimal for Pressure also.
[image]
When I click on the sensor I get this view, there is no decimal for Hum anymore.
[image]

I am using Domoticz and in Domoticz I get one decimal for Temp and no decimal for Hum, maybe this is made by the plugin in Domoticz?

ebaauw, 2020-03-02:
The REST API exposes `state.temperature` in 0.01°C, `state.humidity` in 0.01%, and `state,pressure` in HPa, as per the ZigBee standard for the corresponding ZigBee attributes.  All these are integer values (so 21.00°C is exposed as `2100`).  How to display this is up to the API client, Phoson, Domoticz, or otherwise.

The sensor actually reports pressure in 0.1 HPa using a non-standard attribute, but the REST API plugin doesn't support this (yet?), see #2324.

flopp999, 2020-03-02:
Ok, then I understand why DZ gets the decimals for Temp.

I have not get who develops what.

Have you develop Phoscon?

Smanar, 2020-03-03:
Nope ^^
Phoscon is closed source, most of user you will see here can do nothing on Phoscon.

flopp999, 2020-03-03:
just so I get the background :)

You develop a software called deConz which can communicate with Conbee 1 and Conbee 2?

You also develop the hardware Conbee?

Phoscon is the GUI and is getting it data from deConz?

You have a partnership with Phoscon since you have SD image on your website that includes deConz and Phoscon?

Sorry for all stupid questions 🙈

ebaauw, 2020-03-03:
The ConBee, RaspBee, deCONZ core programme, Phoscon, and SD card image are commercial products by dresden elektronik.  The REST API plugin is open source.  Client apps, like Phoscon, and plugins for home automation systems connect to the REST API.

On this GitHub repository, you'll find the developers who contribute to the REST API plugin.  We're mostly hobbyists, not associated with dresden elektronik.  Some dresden elektronik developers are known to monitor the issues here, but if you want support for the commercial products, best contact dresden elektronik through their support channels.

realwax, 2020-03-06:
@ebaauw Now I understand! Thanks for all your contributions. I thought you all were associated with dresden elektronik.

meinname, 2020-04-22:
Same problem for my Xiaomi Sensors.
But as i have this problem on too many devices i rather replace my ConBee II.
As it worked great for many month on a cheap CC2531.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix