dresden-elektronik/deconz-rest-plugin #2533: Unknown entity when using Xiaomi Smart Plug

alexlopezcifuentes, 2020-03-04:
I have perfectly connected a Xiaomi Smart Plug into Deconz addon. It adds the following entities:

- The actual switch which can be on and off.
- Current power Sensor in Watts (W).
- Another sensor which is in kWh which I assume is power consumption.

However, this last sensor's value is always unknown. Any idea why? Is it really power consumption?

SwoopX, 2020-03-04:
Which deconz version are you using? Can you please provide a screenshots of the basic cluster after the attribute have been read from deconz GUI? We need to understand which modelID it has.

As to my understanding, you get the value "unknown" if no (usable) value is provided. This might be a known issue. You may try the search function with "lumi.plug".

fabsfuchs, 2020-03-11:
I Think I've got a similar problem with the integration of a Xiaomi Smart Plug EU like this one [here](https://www.amazon.de/Xiaomi-Ferngesteuerte-elektronschen-Sprachassistent-Stromverbrauchsmessung/dp/B083F91ZX3/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=Xiaomi+Mi+Smart+Plug&qid=1583957987&s=ce-de&sr=1-1-catcorr).
For me it looks like this:
[image]

[image]

The state.on object is accessible through the REST API so its no problem to toggle the switch, but there is no sensor value created under sensors.

Its possible to read the power value with the deconz-gui under Analog Input (Basic) from Home Automation:
[image]

SwoopX, 2020-03-11:
Oh, great. Another one and completely different than the others...

Regardless, quickly ordered one so I have something to play around with. Small side note: the one above seems to be a Xiaomi Mija. The Xiaomi Aqara ZB3.0 smart plugs seem to be designed better from a zigbee perspective.

SwoopX, 2020-03-13:
@fabsfuchs got the device, but interestingly, I'm missing the last two endpoints which are seen in your screenshot. Looks like that one is going to need some work, so I raise a new device support request for it.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix