dresden-elektronik/deconz-rest-plugin #2534: Coordinator handled as light with 2.05.74

volschin, 2020-03-04:
Updating to 2.05.74 my coordinator is handled as a light and changed its name to "Unknown 8".

I'm using the marthoc docker image.

Smanar, 2020-03-05:
Yep, it' s a new feature
>Support RaspBee, ConBee, ConBee II, XBee under lights resources

But with a little issue > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2525

ACrazyConcept, 2020-03-06:
I actually deleted it from Phoscon thinking it was another orphan device :-O
Looks like no harm was done. It came back on its' own.

wvuyk, 2020-03-09:
Can anyone explain the logic to me why a gateway coordinator (RaspBee, ConBee or a ConBee II) have to emerge as a light? There is nothing to control here? I have to create code now to not use it as a device. 

Trying to wrap my head around it and do not understand a reason of doing this? What are the advantages here?

ebaauw, 2020-03-09:
See https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2472.  The use is limited to setting the name.  The same question applies to exposing the Trådfri repeater as a `/lights` resource (although that one can be made to blink through `state.alert`).

I'm happy to revert creating a `/lights` resource for the coordinator, if it's causing too many issues.  I would like to keep it for other RaspBee/ConBee/ConBee II devices, connected to the network as a router, and for XBee devices.

wvuyk, 2020-03-09:
@ebaauw , Eric

I guess I understand why you would want to see it available like the repeater, so I will put on a filter on it as I really do not need to see these devices in HomeSeer. But would it then be possible to change the type presented in the REST API to a more useful type like "Router", "Coordinator" or "Repeater"?
As the type "Unknown" does worry me as some vendor might present that type for whatever reason? This would give us a clean filter for devices to hide?

ebaauw, 2020-03-09:
> As the type "Unknown" does worry me as some vendor might present that type for whatever reason? This would give us a clean filter for devices to hide?

The `type` is `"Configuration tool"` for the RaspBee/ConBee/ConBee II and `"Range extender"` for the XBee, see #2472.  Unfortunately, something went wrong in merging the various PRs for v2.05.74 (see #2490), causing the `type` to be reported as `"Unknown"`.  This has been fixed in v2.05.75.

wvuyk, 2020-03-09:
Eric,

I upgraded to version 2.05.75 today, but the REST API still shows "Unknown"here. How does one re scan lights that are not lghts? Users are seeing deCONZ cosing when they delete this entry I do think there is something more needed than re-adding it ?

Below the reported REST API with 2.0.5.75?

```
 "29": {
        "etag": "76a5e16f900ab71e4f557c15cc9045f6",
        "hascolor": false,
        "manufacturername": "dresden elektronik",
        "modelid": "RaspBee",
        "name": "Unknown 29",
        "state": {
            "reachable": true
        },
        "swversion": "0x26330500",
        "type": "Unknown",
        "uniqueid": "00:21:2e:ff:ff:01:02:d3-01"
    },
```

ebaauw, 2020-03-09:
> How does one re scan lights that are not lghts?

Just search for lights and read the _Basic_ cluster.  Not sure if that'll update the `type` of the existing resource, through.  You might need to delete the resource first.

> Users are seeing deCONZ closing when they delete this entry.

What do you (they?) mean "deCONZ closing"?  How/where do they delete what entry.

wvuyk, 2020-03-10:
> Just search for lights and read the Basic cluster. Not sure if that'll update the type of the existing resource, through. You might need to delete the resource first.
I do think that most of my users are not used to reading a basic cluster, they are mostly relying on using my plugin to take care of most of the stuff. I am guessing they can use Phoscon eventually.
So I guess I will put the filter on "Unknown" as well then temporary...

> What do you (they?) mean "deCONZ closing"? How/where do they delete what entry.

Two users are reporting that they deleted the light and "both the web page and DeConz crashed and closed. Twice. "  They have reverted to .71.

ebaauw, 2020-03-10:
> So I guess I will put the filter on "Unknown" as well then temporary...

Bad idea, I found the hard way.  Some lights/controllers actually report _Unknown_ as well, see https://github.com/ebaauw/homebridge-hue/issues/640#event-3085726032.

> Two users are reporting that they deleted the light and "both the web page and DeConz crashed and closed. Twice.

Sorry, no clue about that one.  I haven’t had any crash myself.

justme-1968, 2020-03-12:
i would suggest creating this kind of devices (coordinator, repeater) as a sensor. not a light.

rcmcronny, 2020-03-12:
new user, it was also confusing for me, that this is showed as light, i think sensor is really a better kind for this,also.

terfurt, 2020-03-12:
Me to, I'm very confused about the ConBeeII is as an unknown Device in the Phoscon App. First I deleted it as a wrong/ghost Device, later I've seen that is the Stick! I don't understand why it should be shown.

Smanar, 2020-03-12:
For me the "light" type is matches more than "sensor".
If atm you are confused by a new unknow light, why you will not be confused too by an unknow sensor ?

justme-1968, 2020-03-12:
a light is something that illuminates. has an on/off switch and maybe bri and color settings. the support a limited and fixed set of values and commands.

a sensor can be everything else like smoke alarms, thermometers, thermostats, locks, the remote controls, ...

sensors can be read only, can allow for configuration of parameters and can be for invisible things. they are much more flexible in respect to values, commands and configuration.

neither a repeater or the coordinator are lights.

Smanar, 2020-03-12:
The "light" type are not for light, you can find in it fan, wired switch, ect ...
This type is called "light" because of deconz origin, but ATM nothing to see with light.

justme-1968, 2020-03-12:
the 'real' origin of the lights and sensors endpoints is the original hue api from phillips. which i have used to implement the hue/zigbee modules for fhem and deconz has chosen to follow and extend.

i have not come across a single instance where my categorization from above is not correct.

your signal extender example from above is exactly the reason why i have suggested that handling the repeater as a light is also wrong.

the original ikea gateway does not expose the repeater as a light but as its own category. as the hue api knows only light and sensors the repeater belongs to sensors.

SwoopX, 2020-03-12:
Gee, pretty emotional discussion. @ebaauw can speak for himself, but I'd assume he exposed it as light due to the fact that the HW is FFD (routing capabilities, RX on when idle...)

My feeling is that either way (exposing it as light or sensor) would cause confusion. Personally, I'm indifferent if it's there or not since I don't pick it up in FHEM...

justme-1968, 2020-03-16:
this is not really related to fhem but every application that uses the hue api as 'invented' by phillips with the differentiation between the lights and sensors endpoints. which is also not related to routing capabilities. the routing is only related to the power source. normally devices with a constant power supply will route, devices with a battery will not. you can have lights and sensors in both categories. and they will normally route based only on this.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix