dresden-elektronik/deconz-rest-plugin #2539: Feature request: 1) time dependent sensor control, 2) individual scene cycle sets

Kugelfang666, 2020-03-05:
I really like deCONZ running on my home assistant. Currently the system takes care only of my aqara sensors. However, moving forward I'd really like to also move all my hue switches, sensors and bulbs over to deCONZ. Unfortunately, there are two features wich are currently missing for me:

**Time dependent sensor control:**
It would be great if one could specify different scenes to be activated upon motion depending on the time. E.g. 07:00-23:00 Scene A, 23:00-07:00 Scene B

**Scene sets for each switch in one group**
I could not manage to make two dimmer switches within one group cycle through different sets of scenes. Switch 1 Scene 1,2,3 , Switch 2 Scene 3,1,2,6,7

I was about to move my whole setup over but not having a way to specify the scene to be loaded by a sensor as a function of time is deal breaker for me. Currently im using the iOS App iconnecthue to manage my hue bridge which actually allows for both these functions.

Kugelfang666, 2020-03-10:
I just realized that the time dependability of Sensor control has already been requested:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1384

Krocko, 2020-03-11:
You can do all this with the REST API.

Kugelfang666, 2020-03-11:
ok, do you have any starting point for me to find out how?

Kugelfang666, 2020-03-14:
mhm with the app hue essentials this seems to be possible. however, there I cannot set a rule to disable the sensor when lights have been controlled via switch.....

it would be so great to have this function implemented in deCONZ...

zladukas, 2020-03-14:
You can do it with Phoscon, there is manual in https://www.phoscon.de/en/app/doc

Kugelfang666, 2020-03-14:
@zladukas: correct but only if I set the sensor control up with phosncon in which case I have no way to specifically define which scene to activate as a function of time

Phoscon: 
- differentiate sensor control between "Daytime" and "Dusk till Dawn". for me this is too unspecific
- allows to block sensor control on manual (dimmer switch) operation

Hueessentials:
- differentiate sensor control by defining time slots: e.g. 07:00 till 23:00 Scene A, 23:00 till 07:00 Scene B

iConnectHue on Hue Bridge:
- differentiate sensor control by defining time slots: e.g. 07:00 till 23:00 Scene A, 23:00 till 07:00 Scene B
- allows to block sensor control on manual (dimmer switch) operation

zladukas, 2020-03-14:
You can easily change time (or whatever) condition editing deCONZ rules via REST, just find resourcelink, where everything collected: sensors, rules, group